USIWP1Module/USI_I-NUCLEO-LRWAN1 #1: Hardware Documentation?

awneil, 2017-10-20:
Where is the **_hardware_** documentation?

ie, the schematics, pinouts, jumper descriptions, etc?

awneil, 2017-10-21:
Well, user 'Clive One' has posted a pinout list in the ST Community:

https://community.st.com/thread/43664-i-nucleo-lrwan1#comment-172409

But come on, guys - **your customers should _not_ have to be reverse-engineering your products!!**

GrumpyOldPizza, 2017-10-27:
I had e-mailed USI a while ago and gotten no reply. It would be helpful to get the schematics for the WM-SG-SM-42 Shield as an experimentation platform ...

awneil, 2017-10-27:
Yes, I have also emailed them - and had no reply.

Designing-in this module looks like a high-risk venture if this is the level of support they provide ...

:(

GrumpyOldPizza, 2017-10-27:
I would not go that far yet. Perhaps if I'd buy 5000 from them I'd get
attention. But first off I need to evaluate this thing, and they put out
I-NUCLEO-LRWAN1 (or WM-SG-SM-42_Shield_V1.0 as the label says). I cannot
use it properly, because I don't know how to disable the extra stuff on the
PCB to measure power. I cannot check into ANYTHING else but SWD/LPUART,
because I don't know what is connected to what ...

I do kind of like the formfactor, and with a different pinmapping (not
LPUART) it might be interesting as slave to a STM32L4xx (assuming it's
cheaper than the competing module).

USI-ShellyWu, 2018-02-02:
Its schematic has been uploaded at Feb 1, please chech it.

USI-ShellyWu, 2018-02-02:
Please also notice that the latest FW was uploaded.
https://github.com/USILoRaModule/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/wm-sg-sm-42_firmware_v3.3.hex

awneil, 2018-02-02:
Great - a schematic at last!

Thanks.

But why did it have to take over  3 months ??

rogerdahl, 2018-02-02:
Wish it had been only 3 months! It's been 7 months since I picked up my boards at Mouser.

I have been doing reverse engineering and have completed a custom firmware that sends out LoRa packets with values from the sensors. I plan on pushing it to GitHub after some cleanup.

Thanks, though! It's great to have the schematics at last -- better late than never :)

rogerdahl, 2018-02-02:
I also emailed customer support at USI a couple of times asking for this. Never got any reply. That's no way for a company to do business.

USI-ShellyWu, 2018-02-05:
So sorry for missing your requests for long time.

on4eco, 2018-02-11:
Indeed a board that give me headache revers what??

et1975, 2018-02-14:
Are the sources for the shield firmware available somewhere?

USI-ShellyWu, 2018-02-23:
Please check the tree: USI_I-NUCLEO-LRWAN1 / preloaded_firmware / 
v3.3 is the latest

awneil, 2018-02-23:
The request was for **_sources_** - there are only Hex files there.

:( 

https://github.com/USILoRaModule/USI_I-NUCLEO-LRWAN1/tree/master/preloaded_firmware

awneil, 2018-02-26:
You mean: http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-3rd-party-evaluation-tools/i-nucleo-lrwan1.html ?

Does that actually include the source to run **_within_** your module?

USI-ShellyWu, 2018-02-26:
I think it must be built not run within SM42 directly. FW in github can be programmed and run within SM42 that can support AT commands.

closed 2018-07-06 by USI-ShellyWu

bekir35, 2020-08-21:
I have hardware WM-SG-SM-42_SHIELD_V1.0

How can I install the latest version of the usi module on it?

Actually, we controlled from an external host such as NUCLEO-L073 boards.Downloaded the software STM32CubeExpansion_LRWAN_V1.3.1. At_master code works fine but End_Node code is not working.How can we solve this problem?

aithalks, 2021-06-28:
Very Poor documentation by ST Micro electronics.

awneil, 2021-06-28:
@aithalks : it's not ST's product and, therefore, not ST's documentation; USI is a 3rd-party "partner" - so the documentation (or lack thereof) is down to **USI**.

aithalks, 2021-06-28:
@awneil,
But ST's P-NUCLEO-LRWAN2 is bundled with this product.  Is it not your responsibility to make sure that correct document is made available to the user / developer.

awneil, 2021-06-28:
@aithalks I am just a user like you - I have no affiliation with either ST or USI.

But this is USI's GitHub - if you want to raise issues with ST, then you need to do that via ST's own channels.

FWIW, I do agree that the status of "Partner" should imply a greater support / involvement from ST.

aithalks, 2021-06-28:
@awneil : My apologies to you.  After reading your first comment, I got the impression that you are from ST.
Still the bottom line is, showing fingers to other partner, when it comes to solving the problems (Between ST and USI)