USIWP1Module/USI_I-NUCLEO-LRWAN1 #3: sm42_backup doesn't create a backup

OlegSomov, 2018-03-06:
Boar connected to the laptop using UART interface.
The result after running sm42_backup.bat 62 is (62 is the COM port number):
[image]
After that program exits. Tried to connect using SWD connecting and STLink2 but device not even shows up in the Device Manager..

Any ideas?

OlegSomov, 2018-03-06:
Changed COM port number to 28  (so it's between 1 and 60). Same result

USI-ShellyWu, 2018-03-07:
1. Please make sure the comport is configured to be from 1 to 50.
2. Please make sure to run "sm42_backup.bat [comport no.]" as administration, like D:\sm42_fw_update_tool_v1.1\sm42_fw_update_tool_v1.1\bin\sm42_backup.bat 20.

USI-ShellyWu, 2018-03-22:
We have updated WM-SG-SM-42 Update Preloaded AT Command FW Application Note rev. 1.4.pdf for ST-lin/V2 to backup or upgrade. Please follow the latest commands.

closed 2018-07-06 by USI-ShellyWu