USIWP1Module/USI_I-NUCLEO-LRWAN1 #4: Porting to STM32F1xx ?

digidigidigi1, 2018-03-25:
Hi,

Is STM32F103RB & I_Nucleo_LRWAN1 a combination which is supposed to work?

Having the opportunity to experiment with STM32F103RB & I_Nucleo_LRWAN1 I tried to use & port examples from STM32CubeExpansion_LRWAN_V1.1.4 but I got stuck with creation of stm32f1xx_hw_conf.h . I have no idea how to remap those variables. I am using PA2/PA3 (defined as USART2) for modem commands & PC10/PC11 (defined as USART3) for communication with STLink.

I thought that using this 'official' ST&USI libraries path would be easier than for example using arduino path with their support but I am afraid opposite is true.

I understand that not all boards/combinations can be covered in examples but some guidance/tutorial how to port to other boards would be helpful since digging into code has some time limits and at the end of the day it is much cheaper to buy something else & working.

On the bright side I was able to upgrade firmware in I_Nucleo_LRWAN1 or send AT commands via serial port connected to PA2/PA3 pins so at least something works here...

USI-AndyChen, 2018-03-28:
Thank you for your feedback, did you mean you want to create a stm32f1xx_hw_conf.h with the pin mapping from stm32l0xx_hw_conf.h that we provided on the github? the pin mapping is for patch the AT_Slave/End_Node/PingPng application for run on the lora shield board.
Since the PC10/PC11 of STM32F103 seems not SWD/JTAG, so I do not know if they can be used for communicate with ST-LINK.

digidigidigi1, 2018-03-28:
Well,  Since header file 'hw_conf.h' in the example required creation of that stm32f1xx_hw_conf.h  I tried to put it somehow together based on other files in example as well I used some other files from STM32Cube_FW_F1 library.  But at some point I have realized that what was supposed to be fun & learning  turned into  too much digging in code/doc/web & feeling of a waste of time.  

My impression is those examples for LRWAN1 were made quickly probably by ST to have at least something. But they are far from being a tutorial to gain a knowledge for novices. And USI materials didn't made it clearer for me. Yes, there are fixes&schematics but I would say it is mandatory minimum. All together there is a lot of high information on ST site where everything is great and working or low level reference guides for experts which already know anyway.  
It's nothing personal (it's perfect these pages are kept alive) but just the general observation.

Having 2x  I_NUCLEO_LRWAN1 on top STM32F103RB  all I wanted was to try the PingPong example... 

Anyway I learned a lot from this experience. Good luck with the project.

closed 2018-08-02 by USI-ShellyWu