USIWP1Module/USI_I-NUCLEO-LRWAN1 #5: Reading i2c with AT command

Kranichstein, 2018-03-29:
I am trying to read the i2c bus with AT commands, but I am experiencing problems. I have flashed firmware to version 3.6.
I boot up and AT commands work well. Then, I want to read the temp sensor.
According to the datasheet from ST, read address is 0xbf and register 0x0f is a "Who am I"
When I use 
AT+RIIC=0xbf,0x0f,1
the answer is 0, because I did not initialize.
When I initialize with
AT+SIIC=400000,0,0
I do get an OK. However if I send the read command like:
AT+RIIC=0xbf,0x0f,1
the module freezes and I need to reboot.

Can you please hint me towards correct usage of the AT+RIIC command?

Thanks a lot
Andreas

USI-ShellyWu, 2018-04-11:
You may try to read the number at the register firstly, for examples:
# AT+SIIC <CR> //initialize I2C bus in default setting: 400KHz, 7bit address//
OK<CR>
# AT+RIIC=0xbf,0,1 <CR> // I2C address is 0xbf, register address is 0, read length is 1byte. Register 0x00 is a "12".//
0x12 <CR> //get 0x12 from I2C bus//
OK<CR>  //read done//
#

minhngocha, 2018-08-16:
Hi
i'm have problem when using i2c bus.
I initialize i2c bus is AT+SIIC and when  I send the read command 
AT+RIIC=0xBE,32,1 to read  a register,  
the module freezes and I need to reboot.
How to fix problem
thank you

Kranichstein, 2018-08-16:
Look at your board: 
R19 and R21 are not mounted by default. Therefor the module can not access the I2C bus directly. You need to short them.
[image]

minhngocha, 2018-08-16:
thank you, on broad it is r19,r20

awneil, 2019-07-18:
This really should be stated in the User Guide

awneil, 2019-07-18:
> R19 and R21 are not mounted by default. Therefor the module can not access the I2C bus directly

Note that the same applies to other signals from the sensors - Data Ready, Interrupts, etc ...

It looks like the sensors are intended to by used by a "host" microcontroller plugged into the Arduino headers - rather than by the module itself.

rolandvs, 2020-08-16:
Did you check if the required pullup resistors are in the circuit?