USIWP1Module/USI_I-NUCLEO-LRWAN1 #6: Can't use it at all

et1975, 2018-05-02:
`AT`
>OK

`AT+VER`
>ERROR_UNKNOW_COMMAND

Note the spelling on that error!

If I try to flash your latest FW, I get:
>Error: access denied / write protected

So... not only the firmware there is not functional, but it's also write-protected? How do I get around this?

et1975, 2018-05-02:
Starting it with the button pressed unlocks the writes. I was able to flash the new HEX firmware:

>openocd -s scripts \
  -f interface/stlink-v2-1.cfg \
  -f target/stm32l0.cfg -c "init" -c "program wm-sg-sm-42_firmware_v3.6.hex 0x08000000"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v17 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.263937
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x7c205602 msp: 0x3bc81f20
STM32L0: Enabling HSI16
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
adapter speed: 1800 kHz
** Programming Started **
auto erase enabled
Info : Device: STM32L0xx (Cat. 3)
Info : STM32L flash size is 64kb, base address is 0x8000000
Warn : couldn't use loader, falling back to page memory writes
Warn : no flash bank found for address 8010000
wrote 65536 bytes from file wm-sg-sm-42_firmware_v3.6.hex in 8.573833s (7.465 KiB/s)
** Programming Finished **

Now I can't get anything sensible out of the AT interface at all, what does byte `84` mean in response to `AT`?

et1975, 2018-05-03:
I realize now you had a custom version of open-ocd and using the standard version essentially bricks the shield. Trying to use your version after the fact:

>Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : STLINK v2 JTAG v28 API v2 SWIM v17 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.258614
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 3036343262379503
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!

Anything I can do to recover from this?

et1975, 2018-05-07:
Anyone with the working USI and the original FW, what is the correct response to `AT+RMEM=0,16`?

USI-ShellyWu, 2018-05-17:
Could you please check the connections of SWD firstly? Thanks.

et1975, 2018-05-17:
As you can see from the logs, open-ocd can talk to the target programmer, so I think my SWD is ok. The problem is at this point I can't get anything sensible out of AT interface and I didn't create the FW backup.

USI-ShellyWu, 2018-05-23:
I sent the bak file and recovery guide to your mail: eugene@tolmachev.ca

closed 2018-07-06 by USI-ShellyWu

hynci117, 2018-07-15:
Hi, I have a very similar problem with the I-NUCLEO-LRWAN1 - I can't access the target after upgrade:

> "0) stlink-v1"
> "1) stlink_v2"
> "2) stlink_v2-1"
> select a profile no. from above? 2
> going to run using 'stlink_v2-1'
> Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Error: sm42 interface seems abnormal ...
> Info : target responses abnormal ...
> Info : auto-selecting first available session transport "hla_swd". To override u
> se 'transport select <transport>'.
> adapter speed: 300 kHz
> adapter_nsrst_delay: 100
> Info : The selected transport took over low-level target control. The results mi
> ght differ compared to plain JTAG/SWD
> none separate
> Info : Unable to match requested speed 300 kHz, using 240 kHz
> Info : Unable to match requested speed 300 kHz, using 240 kHz
> Info : clock speed 240 kHz
> Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
> Info : using stlink api v2
> Info : Target voltage: 3.252071
> Error: init mode failed (unable to connect to the target)
> in procedure 'init'
> in procedure 'ocd_bouncer'
> 

Can you please post the recovery procedure?

USI-ShellyWu, 2018-07-16:
Did you backup the FW firstly?

hynci117, 2018-07-16:
No, unfortunately the process failed and I have no backup. I can access the mcu via stlink, but that is all.

hynci117, 2018-07-16:
Yes, my fault. I didn't realize the backup didn't run and proceeded. 
The Device ID seen by STLink utility is 0x417

USI-ShellyWu, 2018-07-17:
I marked by the bold "Info : found unique device id = 3036343262379503"

Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select '.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : STLINK v2 JTAG v28 API v2 SWIM v17 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.258614
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
**Info : found unique device id = 3036343262379503**
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!

hynci117, 2018-07-17:
The ocd says it can not connect to the target and so it is not displaying the unique device id as in your example. The output for me ends like this:
> ...
> Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
> Info : using stlink api v2
> Info : Target voltage: 3.252071
> Error: init mode failed (unable to connect to the target)
> in procedure 'init'
> in procedure 'ocd_bouncer'
I have two modules, the other one is fine. Can I use the backup from the second one?

USI-ShellyWu, 2018-07-17:
Yes, backup firstly. Then keep power on and upgrade FW.