USIWP1Module/USI_I-NUCLEO-LRWAN1 #7: problem with patch

borombo96, 2018-06-28:
I download the patch file of I CUBE LRWAN 1.1.5
When write the command to patch it give me this error:

$ patch p1 < sm42-lrwan_v1.1.5.patch
patching file p1
Hunk #1 FAILED at 138.
patch: **** Can't reopen file p1 : No such file or directory

Someone can help me?

USI-ShellyWu, 2018-07-03:
Please correct the command "$patch -p1 < sm42-lrwan_v1.1.5.patch", and notice that execute the command under tha patch "  .\STM32CubeExpansion_LRWAN_V1.1.5\  "

borombo96, 2018-07-03:
I've patched it with a Linux SO, but it doesn't work... I'm try the end node example with a LoRa gateway, which parameters I have to modify in the file commisioning.h?  I have to follow this instruction from Readme? 
"a.	in 'en.i-cube_lrwan\STM32CubeExpansion_LRWAN_V1.1.0\Projects\Multi\Applications\LoRa\End_Node\inc\Comissioning.h':

// correct the definition below depended on your sx1301 gateway:

OVER_THE_AIR_ACTIVATION

LORAWAN_APPLICATION_EUI

LORAWAN_APPLICATION_KEY

LORAWAN_DEVICE_ADDRESS

LORAWAN_NWKSKEY

LORAWAN_APPSKEY

b.	In 'en.i-cube_lrwan\STM32CubeExpansion_LRWAN_V1.1.0\Middlewares\Third_Party\Lora\Mac\region\RegionEU868.h'

#define EU868_DUTY_CYCLE_ENABLED 0 // to enabling to update node data every 10 seconds to gateway

c.	In 'en.i-cube_lrwan\STM32CubeExpansion_LRWAN_V1.1.0\Middlewares\Third_Party\Lora\Core\lora.c'

#define USE_SEMTECH_DEFAULT_CHANNEL_LINEUP 1 // let the node uses DR3 on RX2
" 
Thank you for help.

USI-ShellyWu, 2018-07-05:
Please refer to the setting: https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/tree/master/I-CUBE-LRWAN_patchs/I-CUBE-LRWAN%201.1.5
Also, according to your gateway please set DevEUI, DevAddr, NSK, ASK.... to the correct values.

MalekT, 2018-07-05:
I have the same problem with patch also (Hunk #1 FAILED at xxx (Different line endings) ).
I think that that error is caused by the difference in line endings between files. 
The line endings are CRLF (\r\n) instead of being LF (\n) only, as line endings are non-visible, I can't see anything.
So how can i fix that problem please ?

USI-ShellyWu, 2018-07-06:
Our patched had been built in Cygwin run in Windows. The problem you mentioned should be caused by different OS. If you patch in linux, please firstly modify the line ending in dos2unix run in Linux, then try to patch again.

MalekT, 2018-07-06:
Thank you for your reply !
I'm patching using Git on Windows.

closed 2018-10-16 by USI-ShellyWu