USIWP1Module/USI_I-NUCLEO-LRWAN1 #8: Patch issue

MalekT, 2018-07-05:
When applying (manually) the patch manually i noticed some problems that confused me: 

 in " ../Projects/Multi/Applications/LoRa/End_Node/inc/stm32l0xx_hw_conf.h " : 
1) there is those lines:
+#define RADIO_ANT_SWITCH_PORT                     GPIOA
+#define RADIO_ANT_SWITCH_PIN                      GPIO_PIN_4

 #define BAT_LEVEL_PORT                            GPIOA
 #define BAT_LEVEL_PIN                             GPIO_PIN_4

is it normal to define  two different peripherals using the same pin (PA4) ? 

2)
-#define USARTX_TX_PIN                  GPIO_PIN_2
+#define USARTX_TX_PIN                  GPIO_PIN_14
....
....
-#define USARTX_RX_PIN                  GPIO_PIN_3
+#define USARTX_RX_PIN                  GPIO_PIN_1

this configuration generated debugging problem so i checked the stm32L053R8 pinout configuration and  i found that the PA3 alternate is PA15 and not PA1 (PA2 alternate is PA14).

What do you think ?

USI-ShellyWu, 2018-07-06:
For question1, the patch released for 1.1.5 does not contain this Battery Level GPIO. In this patch, Battery function GPIO is following
#define BAT_LEVEL_PORT                            GPIOA
#define BAT_LEVEL_PIN                             GPIO_PIN_7

For question2, in USI IO header file there defined UART to output debug log, not used to data transmit. TX is PA14

MalekT, 2018-07-06:
Alright ! Thanks for your reply ! 
Those configurations (question 2.) are in the stm32l0xx hardware configuration header file. 
I'm running the EndNode application using an stm32l073 and I-nucleo-LRWAN1 shield but the application doesn't work until i reconfigure PA2 defining USARTX_TX_PIN GPIO and PA3 defining USARTX_RX_PIN GPIO.
However, in all cases the EndNode application doesn't work for this board (there is no connection with the gateway). 
The application works using the B-L072Z-LRWAN1 board that's why i think the problem is caused by the I-nucleo-lrwan1 shield.

USI-ShellyWu, 2018-07-06:
It is seemed to be solved, right?

MalekT, 2018-07-06:
Not yet unfortunately.
When running the EndNode application using NUCLEO-L073RZ with I-Nucleo-LRWAN1 shield in the hyperterminal i get only the gateway activation keys but no transmitted data to the network server and my device still not connected to the Gateway. 
Ps: it is working without problems when i change the board , i used the B-L072Z-LRWAN1 board. So what is the problem with the NUCLEO-L073RZ and its I-Nucleo-LRWAN1 shield?

USI-ShellyWu, 2018-07-09:
Please tell the steps and commands for using NUCLEO-L073RZ with I-Nucleo-LRWAN1 shield when running the EndNode application.

MalekT, 2018-07-09:
I flashed the EndNode application code from the STM32CubeExpansion_LRWAN_V1.1.5 in my NUCLEO-L073RZ with I-Nucleo-LRWAN1 shield, I only disabled the Low power mode and did nothing else into the code. (I'm using µvision keil to build and run the program).
I patched the files  of the STM32CubeExpansion_LRWAN_V1.1.5 using your patch file. 

Then when running the program, on the hyperterminal i only get the devEui , the AppKey and AppEui values but never connected to the the Gateway and the network server.
 
I am using the WM-SG-SM-42_Shield_V1.0  and i didn't apply the firmware updates ( i don't know which firmware update to apply).

USI-ShellyWu, 2018-07-10:
Please provide the fail log.

MalekT, 2018-07-10:
[NUCLEO_CUBE_LORA.build_log.pdf](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/2179779/NUCLEO_CUBE_LORA.build_log.pdf)

MalekT, 2018-07-10:
Here is my log file: 

[NUCLEO_CUBE_LORA.build_log.txt](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/2179839/NUCLEO_CUBE_LORA.build_log.txt)

MalekT, 2018-07-11:
This is what is happening  in the hyperterminal , the Tx and Rx are never done , the network server is never joined. 
[image]

MalekT, 2018-07-12:
So finally I have the answers for all the problems and I arrived to establish the connection between my Nucleo L073 board+I-Nucleo_LRWAN1 shield and the Network server.

Well, it is NOT possible to execute the End-Node example on the Nucleo L073 board+I-Nucleo_LRWAN1 shield: the End-Node application drives a SX12xx radio shield and cannot drive a LoRa modem.

So I loaded the AT_Master example (available in the STM32CubeExpansion_LRWAN_V1.1.5) in Nucleo L073 board+I-Nucleo_LRWAN1 shield with some hardware and software configurations (in theNucleo L073 board: SB13, SB14 should be unsoldered  and SB62 ,SB63 should be soldered) and using the USART2. 
We should also use I-Nucleo_LRWAN1 DevEui.

Thanks for your replies :-)

USI-ShellyWu, 2018-07-13:
Good news

closed 2018-07-13 by MalekT

Abdoul2019, 2019-04-02:
@MalekT I have the same issue with my Nucleo L073 board+I-Nucleo_LRWAN1 shield. Can you give more details how you solve the problem.
Thanks

VelcoH, 2019-06-14:
@MalekT @USI-ShellyWu, how were you able to solve this? I also have this kind of a problem.