USIWP1Module/USI_I-NUCLEO-LRWAN1 #9: Cannot update firmware

baptiste-cartier, 2018-07-16:
Hello,

I am trying to update from firmware 2.6 to 3.7 (following the document https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20Update%20Preloaded%20AT%20Command%20FW%20Application%20Note%20rev.%201.4.pdf) , but whatever I try, I get this : 

`PS D:\sm42_fw_update_tool_v1.1-1\bin> .\sm42_backup.bat 6
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 0
going to run using 'stlink-v1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
PS D:\sm42_fw_update_tool_v1.1-1\bin> .\sm42_backup.bat 6
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
PS D:\sm42_fw_update_tool_v1.1-1\bin> .\sm42_backup.bat 6
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...`

I uninstalled every driver and installed their newer version, ST-Link utility is up to date, the st-link driver is up to date. When trying to access the STM32L0 on the board via ST-Link, it works fine : 

`11:08:39 : ST-LINK SN : 52FF71066765565248570467
11:08:39 : ST-LINK Firmware version : V2J29S7
11:08:39 : Connected via SWD.
11:08:39 : SWD Frequency = 4,0 MHz.
11:08:39 : Connection mode : Connect Under Reset.
11:08:39 : Debug in Low Power mode enabled.
11:08:39 : Device ID:0x417 
11:08:39 : Device flash Size : 64KBytes
11:08:39 : Device family :STM32L0xx 64k
`
ST-LINK detects the 3.7 hex as corrupted, I already downloaded it multiple times and it did not change.

If you have any ideas as to why I can't update the firmware that'll be of great help

baptiste-cartier, 2018-07-17:
It appears that trying to backup the firmware erased the existing firmware. Now the I-NUCLEO-LRWAN1 does not respond to any AT commands and ST-LINK shows that the memory has been wiped clean (only zeros). My module is now unusable, is there any way to fix this ? And what could have gone wrong in the backup process ?

baptiste-cartier, 2018-07-18:
Here is the device ID when I try to upload a new firmware : 
"found unique device id = 3438343369368208"

USI-ShellyWu, 2018-07-19:
I will send your the recover guide, please kindly give your mail.

baptiste-cartier, 2018-07-19:
Here is my email : baptiste.cartier.job@gmail.com

tommasopolonelli, 2018-07-25:
I have the same problem, could you send me the recover guide ?

My email is: tommyp1990@libero.it

USI-ShellyWu, 2018-07-26:
It needs your device ID as above.

tommasopolonelli, 2018-07-26:
I cannot read the ID because the module's flash is erased, I tried to load a custom firmware with STLink unsuccessfully... and now the module is completely useless...

USI-ShellyWu, 2018-07-27:
We have to need device ID to recover module. Without it, there does not have another way to recover.

hynci117, 2018-07-28:
Hi, I am still struggling with the update as well (same command line outputs as those listed above). I have tried updating 2 units without success. One does not even connect with the update tool (no unique id listed, just error Error: sm42 interface seems abnormal ...). The other unit shows its number (found unique device id = 3438343367367F08) and works fine with the AT commands, but it is not possible to backup or update as well. Both units are accessible via STLink, so it should not be a wiring problem.
Is it possible to recover the first unit (with unknown id)? Would you send me the "secret" recovery guide? My mail: hynci317@gmail.com

USI-ShellyWu, 2018-08-01:
May I know that what FW you have updated and the detailed steps for updating?

hynci117, 2018-08-02:
Ok, my steps are as follows:
- the device fw is V2.6 (using command "ATI")
- new firmware I am trying to update is V3.7
- update tool is sm42_fw_update_tool_v1.1-1
- STLink version is V2.J29.M18, using COM5
- running as administrator on Win7

Trying to backup:
> D:\Lora\fw_update_tool\sm42_fw_update_tool_v1.1-1\bin>sm42_backup.bat 5
> "0) stlink-v1"
> "1) stlink_v2"
> "2) stlink_v2-1"
> select a profile no. from above? 2
> going to run using 'stlink_v2-1'
> Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Error: sm42 interface seems abnormal ...
> D:\Lora\fw_update_tool\sm42_fw_update_tool_v1.1-1\

Trying to update:
> D:\Lora\fw_update_tool\sm42_fw_update_tool_v1.1-1\bin>sm42_update.bat 5 wm-sg-sm-42_firmware_v3.7.hex
> "0) stlink-v1"
> "1) stlink_v2"
> "2) stlink_v2-1"
> select a profile no. from above? 2
> going to run using 'stlink_v2-1'
> Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Error: sm42 interface seems abnormal ...
> Info : target responses abnormal ...
> Info : auto-selecting first available session transport "hla_swd". To override u
> se 'transport select <transport>'.
> adapter speed: 300 kHz
> adapter_nsrst_delay: 100
> Info : The selected transport took over low-level target control. The results mi
> ght differ compared to plain JTAG/SWD
> none separate
> Info : Unable to match requested speed 300 kHz, using 240 kHz
> Info : Unable to match requested speed 300 kHz, using 240 kHz
> Info : clock speed 240 kHz
> Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
> Info : using stlink api v2
> Info : Target voltage: 3.253088
> Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
> Info : STM32L flash size is 64kb, base address is 0x8000000
> Info : found unique device id = 3438343367367F08
> Info : atcmd responses abnormal, will use previous nvm backup
> Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECO
> VERY by this tool !!!
> 
> D:\Lora\fw_update_tool\sm42_fw_update_tool_v1.1-1\bin>

hynci117, 2018-08-03:
Ok, my steps are as follows:
- device fw is V2.6 (using command "ATI")
- new firmware I am trying to update is V3.7
- update tool is sm42_fw_update_tool_v1.1-1
- STLink version is V2.J29.M18, using COM5
- running as administrator on Win7 64bit:

D:\Lora\fw_update_tool\sm42_fw_update_tool_v1.1-1\bin>sm42_backup.bat 5
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
D:\Lora\fw_update_tool\sm42_fw_update_tool_v1.1-1\

closed 2018-08-07 by USI-ShellyWu

hynci117, 2018-08-07:
So is there a solution? The issue is now closed with now solution provided...

USI-ShellyWu, 2018-08-08:
The bak file was sent to your gmail yesterday and the recovering guide have been uploaded in Github: https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/tree/master/preloaded_firmware, also mentioned in the mail to you.

hynci117, 2018-08-10:
OK, thanks. This finally did it.
However it needs some clarification. Maybe for others:
When updating the I-NUCLEO-LRWAN1 shield there is a little difference. As the update tool prompts you to disconnect swd and hit enter to continue - this wont work as the port number in the command was the  stlink and the tool won't be able to access the module. So I had to disconnect the stlink and then change the port number of the module serial to be the same as was the stlink port.

> sm42_update.bat 5 wm-sg-sm-42_firmware_v3.7.hex  _<-- here "5" is my stlink port number_
> .....
> DO NOT TERMINATE the process or the window.
> to ensure target is in running state, please unplug SWD wire and REPOWER target
> 
> then please hit ENTER key while ready to continue ....
> _--> now disconnect stlink from target and unplug stlink. reset the target, in serial terminal check the AT response of the module - in my case this was port "4":_
> _the module response in the terminal to AT was ERROR_INVALID_MODULE  <-- this seems to be okay_
> _--> now close the serial terminal and change the module serial port number "4" to "5"  (the previous stlink port number)_
> _--> go back to the command prompt with the update tool and hit enter_ 
> Info : restoring nvm ...
> ................................................................................
> fw updated successfully!please takes a reset for effect the new fw.
> 

So after two weeks I have my module updated. Hopefuly the rest of the modules will go smoother. But still it is a really painful procedure, not quite suitable for a production environment. Is there a faster way to upgrade the modules in bulk?

posyp123, 2019-06-06:
Hi, I have the same problem. My module id = 3438343369368309 and my mail is pawelprogram@gmail.com 
Could you send me the recover guide too?

USI-ShellyWu, 2019-06-10:
Hi, recover guide can be downloaded from https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20recover%20guide.pdf.

posyp123, 2019-06-10:
I have try this guide but without bak file I can not execute it fully. Am I right that I need that file?

USI-ShellyWu, 2019-06-17:
the bak file has been sent to you last week, please check in your gmail.

dominon, 2019-08-28:
Hi I got two new boards I-NUCLEO-LRWAN1 and on one AT commands does not work. It seems that it came with corrupted firmware? I can connect to ST micro in module and I was able to read unique ID: 303634326A379603 if I'm correct USI-ShellyWu with .bak file generated from you I can bring this module back to operational? If yes can you send me one at pacminik@gmail.com
Thank you in advance.

USI-ShellyWu, 2019-08-29:
Please kindly send the log that AT commands does not work, and you should also confirm the hardware of the I-NUCLEO-LRWAN1 is all right.
BAK file is useful when the board cannot update FW, You can see the log sent by hynci117 on 2018/Aug/03 that shows the board cannot update FW needs to be recovered with BAK file.

dominon, 2019-08-29:
When I try to send AT commands, module it is not responding at all so I tried to update FW starting from backup and I got below. I did thesame on second module and it made backup no problems. Below log from not working one.

`Microsoft Windows [Version 10.0.17134.885]
(c) 2018 Microsoft Corporation. Wszelkie prawa zastrzeżone.

C:\sm42_fw_update_tool_v1.1-1\bin>sm42_backup.bat 14
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...

--> Here I wanted to force update but I end up with no .BAK file. 

C:\sm42_fw_update_tool_v1.1-1\bin>sm42_update.bat 14 wm-sg-sm42_firmware_v3.7.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.251410
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 303634326A379603
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!

C:\sm42_fw_update_tool_v1.1-1\bin>`

Working board have firmware version 2.9 and since they were bouhgt together in one bath they should be thesame. So as I understand since my AT commands doesn't work and I cannot backup\update I need BAK file from you to force update,  so I at least end up with known working firmware version.

USI-ShellyWu, 2019-08-29:
The bak file will be sent to you late, please try it for recovery.

AnTrii, 2019-09-05:
Unable to make a backup and firmware update on two I-NUCLEO-LRWAN1 modules. UDIDs: 303634324F376C02 and 3438343385368E09. Tried to use STLink built in two Nucleo boards: NUCLEO-L476RG and NUCLEO-L152RE, both have firmware V2J34M25. @USI-ShellyWu, please send me a BAK files for my modules. E-mail: antrii@mail.ru. Command line logs are listed below.
Thank you in advance.

***
ST-LINK on NUCLEO-L476RG, UDID 303634324F376C02:
```
D:\MCU\ST\Nucleo\_EXPANSION\I-NUCLEO-LRWAN1\Firmware\sm42_fw_update_tool_v1.1-1\bin>sm42_backup.bat 12
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
D:\MCU\ST\Nucleo\_EXPANSION\I-NUCLEO-LRWAN1\Firmware\sm42_fw_update_tool_v1.1-1\bin>sm42_update.bat 12 wm-sg-sm-42_firmware_v3.3.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : STLINK v2 JTAG v34 API v2 SWIM v25 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.261983
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 303634324F376C02
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!
```

ST-LINK on NUCLEO-L152RE, UDID 3438343385368E09:
```
D:\MCU\ST\Nucleo\_EXPANSION\I-NUCLEO-LRWAN1\Firmware\sm42_fw_update_tool_v1.1-1\bin>sm42_backup.bat 4
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
D:\MCU\ST\Nucleo\_EXPANSION\I-NUCLEO-LRWAN1\Firmware\sm42_fw_update_tool_v1.1-1\bin>sm42_update.bat 4 wm-sg-sm-42_firmware_v3.3.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : STLINK v2 JTAG v34 API v2 SWIM v25 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.245473
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 3438343385368E09
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!
```

rosettapagliuca, 2019-11-24:
Hi, I have the same problem as above. I am trying to update the version v3.7.
I am using the ST-Link on Nucleo L152RE and the device id=0700F8FF92516DAE

This is the output of update:
`

```
command: sm42_update.bat 4 wm-sg-sm-42_firmware_v3.7.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v34 API v2 SWIM v25 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.226931
Info : stm32l1.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : STM32L flash has dual banks. Bank (0) size is 256kb, base address is 0x8000000
Info : found unique device id = 0700F8FF92516DAE
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!`
```
My email is leonardo.severi@libero.it

USI-ShellyWu, 2019-11-28:
Done, the bak file has been sent to you.

massroukimohamedtaher, 2020-03-02:
Hello I have the same trouble ^^ please help me :D

1hwg, 2020-09-26:
Backup.bat will not be performed correctly (sm42 interface seems abnormal ...) if ECHO is enabled in the WM-SG-SM-42 module.
You must prefix backup.bat with the following sequence of AT commands from the terminal.

ATE = 0 / * disable the local echo * /
AT + WDCT / * save changes to eeprom * /

then the module power cycle.

then backup.bat

jzorroperez, 2020-10-28:
Hi, I tried to solve the problem with ATE = 0 ans AT+WDCT but i don't have a correct answer with "0) stlink-v1"; "1) stlink_v2" or "2) stlink_v2-1".
@USI-ShellyWu May you send me a backup of my +EUI=e2,4f,43,ff,fe,44,ce,f3?
```
C:\Users\jzorr\Downloads\SetProgramas\sm42_fw_update_tool_v1.1-1\sm42_fw_update_tool_v1.1-1\bin>sm42_backup.bat 6

0) stlink-v1
1) stlink_v2
2) stlink_v2-1
select a profile no. from above? 1
going to run using 'stlink_v2
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : starting to backup nvm ...
................................................................................................................................................................................................................................................................
Info : backup nvm is valid
Info : current fw version in module:  1.100000
Info : found eui: E24F43FFFE44CEF3
Info : finish nvm backup.
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Error: open failed
in procedure 'init'
in procedure 'ocd_bouncer
```

ANauzet, 2020-11-17:
Hello,
I'm trying to backup the firmware in my to module, without succes so far.
@USI-ShellyWu , could you please send me the two bak file for the following module.
My adresse: antoine.nauzet@gmail.com

DevEUI: 0xe24f43fffe44ccdf
DevEUI: 0xe24f43fffe44d1ce

Thank you

ANauzet, 2020-11-17:
Hello,
I could not make backup working, even with the "echo off" command. I still have the "sm42 interface seems abnormal"
What does it mean?

I have asked for the .bak file, but will the update work? I do not want to have a module not working because update crash.

Thank you

ugrakm, 2020-11-21:
Hello,
I'm trying to backup the firmware in my to module, without succes so far.
@USI-ShellyWu , could you please send me the  bak file for the following module.
My adresse: ugrakm@gmail.com

[image]

Thank you

ugrakm, 2020-11-21:
Hello,
I'm trying to backup the firmware in my to module, without succes so far.
@USI-ShellyWu , could you please send me the bak file for the DevEUI:3438343388367E08
My adresse: ugrakm@gmail.com

[image]

ANauzet, 2020-11-26:
Hello,
Is there anyone from USI that could support me? I steel need tthe backup file for my module. And no response.
Please help
Thank you

ugrakm, 2020-12-10:
I have asked for the .bak file, But Unfortunately No one has not been hear my S.O.S message. I steel need the backup file for my module. And no response .
**Could you please send me the *.bak file for the Device ID 3438343388367E08**
My adresse: ugrakm@gmail.com

ANauzet, 2020-12-10:
Unfortunately, no one from USI is answering anymore here.
I even try the contact page on there web site, and no more answer there!