USIWP1Module/USI_I-NUCLEO-LRWAN1 #10: Can not get a buffer bigger than 4 bytes at the reception

adrienDLCT, 2018-07-25:
Hello,

I am working with a stm32l053R8 and a I-NUCLEO-LRWAN1.
I've got a problem with my buffer at the reception in fact when i read it, every time i see only the first 4 bytes and after nothing. So I verified if the data are well transmitted and indeed on the TX-Uart pin of the LoRa shield i can read in Hexa that the message is correct. So the problem come from the function that read these data an put it in the buffer; LoraRadio.read() 
If you have an idea how to solve this problem, I would be greatfull to have the solution.
Thank you.

USI-ShellyWu, 2018-07-26:
Please kindly tell what function you verified and detailed step and the final fail log.

adrienDLCT, 2018-07-26:
uint8_t rcvData[20];  
  bool next = true;
  if ( init_buff == true)
  {
    for (int i=0; i<20; i++)
    {
      rcvData[i] = '0';
    }
    Serial.print("buffer initialized");
    init_buff = false;
  }
 if(loraRadio.read(rcvData) > 0 && next == true)
  {
    //loraRadio.read(rcvData);
    Serial.print("buffer : "); 
    Serial.println((char*)rcvData);
    next = false;
}

This code display me the 4 first bytes received, and after there are the '0' I put to initialize the buffer.

PS: When I am watching the information received on the TX pin of the LoRa shield with a logic analizer, i can see the full message of 20 bytes.

USI-ShellyWu, 2018-08-02:
As talked with other engineer, sorry for not understanding your problem. We must know the function you tested and the detailed test steps and the modification you have made before.

adrienDLCT, 2018-08-02:
So, this is the entire code, I want to send the data sensore in line in the buffer and then at the reception, separate everything, but at the reception the buffer is badly fill.
Sorry if my explanations are not clear to you.

**#include <HTS221Sensor.h>                                       // Library of temperature and humidity sensor
#include "LoRaRadio.h"
#include "lora_driver.h"

//Define
#define I2C2_SCL    PB8                                         // Define of I2C clock pin
#define I2C2_SDA    PB9                                         // Define of I2C data pin
//#define gateway                                                 // Here we define the Node we want to program Master or Slave
#define node1
//#define node2

HardwareSerial SerialLora(PC_11, PC_10); 

HTS221Sensor  *HumTemp;
TwoWire *dev_i2c;
uint8_t Msg1[] = "temp";                                   //Declaration of humidity message
uint8_t Msg2[] = "humi";                                   //Declaration of temperature message
uint8_t Node1[] = "A";                                      // ID Node 1 size = 2 bytes
uint8_t Node2[] = "B";                                      //ID Node 2  size = 2 bytes

void ShiftLeft(uint8_t Oldbyte[], uint8_t Newbyte[], int nbr_shift, int size_oldB)
{
  for (int i = 0; i<4; i++)
    {
      Newbyte[i]=Oldbyte[i+nbr_shift];
    }
}
void ShiftRight(uint8_t Oldbyte[], uint8_t Newbyte[], int nbr_shift, int size_oldB)
{
  for (int i = 0; i<size_oldB; i++)
    {
      Newbyte[i+nbr_shift]=Oldbyte[i];
    }
}

void readbuffer(uint8_t rcvData[64], bool next)               // Function that read data of the buffer
{
    Serial.print("dans fonction buffer \n"); 
    
    uint8_t Sens_temp[5];
    uint8_t Data_sens_temp[4];
    uint8_t Sens_humi[5];
    uint8_t Data_sens_humi[4];
    
    for(int i=1; i<6; i++) 
    {
      Sens_temp[i-1] = rcvData[i];
    }
    Serial.println((char *)Sens_temp);
    for(int i=5; i<10; i++)
    {
       Data_sens_temp[i-5] = rcvData[i];
    }
    float data = (float)*Data_sens_temp;
    Serial.println(data);
    
    for(int i=10; i<15; i++) 
    {
      Sens_humi[i-10] = rcvData[i];
    }
    Sens_humi[4] = '\0';
    Serial.println((char *)Sens_humi);
    for(int i=15; i<19; i++)
    {
       Data_sens_humi[i-15] = rcvData[i];
    }
    data = (float)*Data_sens_humi;
[... 115 lines cut ...]
      Serial.println((char)LoraRadio.read());
    }*/
  if(loraRadio.read(rcvData) > 0 && next == true)
  {
    loraRadio.read(rcvData);
    Serial.print("buffer : "); 
    Serial.println((char*)rcvData);
    Serial.println((char)rcvData[4]);
    digitalWrite(LED_BUILTIN, HIGH);                            // Turn on the internal LED
    if((char)rcvData[0] == 'A')
    {
      Serial.print("data node 1 : ");                                     // Send on serial port Hum:
      readbuffer( rcvData, next);                                         // cf : readbuffer function
      next = false;
    }
    if((char)rcvData[0] == 'B')
    {
      Serial.print("data node 2 : ");                                     // Send on serial port Hum:
      readbuffer( rcvData, next);                                         // cf : readbuffer function
      next = false;
    }
  }
    digitalWrite(LED_BUILTIN, LOW);                               // Turn on the internal LED
}
#endif**

USI-ShellyWu, 2018-08-07:
Please try to configure UART to be 115200 firstly, and try USI FW supports AT CMD to confirm work normally.

cstratton, 2020-11-12:
Likely your loraRadio.read(rcvData) method is faulty but since you have not provided it no one can really know.

You also appear to be calling it twice which is probably a mistake...