USIWP1Module/USI_I-NUCLEO-LRWAN1 #11: Saving to EEPROM not working

hynci117, 2018-08-31:
The problem:  when the module is in verbose mode, saving address with the WDCT command will not work. Firmware version 3.7
Command sequence example:
> ATZ
> \# đ                                                                  
> AT+VERB=1
> \# OK                                                                    
> AT+DEFMODE=0
> \# OK                                                                 
> AT+ADDR=11223344
> OK                                                              
> \# 
> AT+WDCT                                                              
> \# OK
> ATZ
> \# đ                                                                
> AT+ADDR
> \# +ADDR=00,00,00,00
> OK                                                              
> \# 

As you can see, after reset the address is set to zeroes.
Then when I clear the settings with AT+WDCT=0 command and repeat the same sequence **without** setting verbose output to 1, everything works as expected and the address is kept in eeprom. 
Is there a workaround to to use both verbose outputs and the ability to persist settings in eeprom?

USI-ShellyWu, 2018-09-04:
Please try below after modified address:
AT+ADDR= 11223344
AT+WDCT
AT+ADDR //the address is 11 22 33 44
AT+DEFMODE=6
ATZ // reset shield board
AT+DEFMODE // sure that mode is lora wan mode
AT+ADDR  // the address should be 11 22 33 44

USI-ShellyWu, 2018-09-06:
the address is valid in lora wan mode, not for other modes.

closed 2018-09-10 by USI-ShellyWu