USIWP1Module/USI_I-NUCLEO-LRWAN1 #12: Impossible to update firmware

Manitou02, 2018-09-03:
Hi, 

I contact because I have tried to update the LRWAN USI module. I have respected the procedure inside pdf document but now the module seems to be empty, no AT command responds.
The ST link is working because I can connect on the chip but Read protected and no data access...
SWD ST LINK V2 is updated with the last version

First: backup command 
C:\USI\bin>sm42_backup.bat 0
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 2.982295
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target state: halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0xf1000003 pc: 0xfffffffe msp: 00000000
STM32L0: Enabling HSI16
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
adapter speed: 1800 kHz
** Programming Started **
auto erase enabled
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 3438343368369508
Info : nvm was backup in '3438343368369508.bak'
** Programming Finished **
** Resetting Target **
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
shutdown command invoked

---------------------
And now update : with error messages.

C:\USI\bin>sm42_update.bat 0 wm-sg-sm-42_firmware_v3.7.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 2.571541
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 3438343368369508
Info : nvm was backup in '3438343368369508.bak'
Info : stm32lx_write_rdp RDP=aa...
Info : stm32lx_write_rdp OBL_LAUNCH...
STM32Lx unlocked, wait 1 sec.
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target state: halted
target halted due to debug-request, current mode: Thread
xPSR: 0xf0000000 pc: 00000000 msp: 00000000
STM32L0: Enabling HSI16
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
adapter speed: 1800 kHz
** Programming Started **
auto erase enabled
Info : loading file 'wm-sg-sm-42_firmware_v3.7.hex' ...
Info : loaded 181041 of 181041
found fw version 0 in 'wm-sg-sm-42_firmware_v3.7.hex'
Info : going erase & flash ...
Warn : couldn't use loader, falling back to page memory writes
wrote 65536 bytes from file wm-sg-sm-42_firmware_v3.7.hex in 24.066284s (2.659 KiB/s)
Info : stm32lx_write_rdp RDP=0...
Info : stm32lx_write_rdp OBL_LAUNCH...
driver protected
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
** Programming Finished **
** Resetting Target **
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
shutdown command invoked
      0 [main] openocd-dev 7952 cygwin_exception::open_stackdumpfile: Dumping stack trace to openocd-dev.exe.stackdump

Thank you for your support
BR
Manitou

USI-ShellyWu, 2018-09-04:
Please refer to https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/README.md. 
"!!! Important Note !!!

since the shield board with the preloaded firmware has Flash read protection, it is necessary to backup whole flash/eeprom before any flash programming, please read recover guide in the document below before any flash operations:"

I will send you bak file later on.

USI-ShellyWu, 2018-09-10:
Please kindly tell your mail contact.

USI-ShellyWu, 2018-09-10:
Or you can download from Baidu Cloud
https://pan.baidu.com/s/1u0QM--3EHKnjuT-OfgCXFA 
password: pefr

The link will be valid in 7 days.

Manitou02, 2018-09-10:
Hello,
Thank you for your support.
Sorry for my late answer but too busy this week. I try to solve my problem for the next week.
Thank you for the link, the bak file is correctly downloded.
I keep you informed.

lukblt, 2018-10-06:
Hello,
i have the same issues, saved the bak and tryed to update firmware... did not work :(

lukblt, 2018-10-08:
Hello, 

i connected the Hardware like in your "Update Preloaded...."-Doc.

9.1 Backup NV data worked like you discirbed and i got my backup .

In the "Recovery Guid" - 
"When you see this information: “then please hit ENTER key while ready to continue ....”, Disconnect the EVB power supply to remove st-link and then re power the EVB."

after this step the Controller crashed and now i cant conect to the Controller again (debug dead?)

Cant you do a good Document for your Product? Several People had Problems with your Firmware-Flashing Methode.

USI-ShellyWu, 2018-10-09:
For backup and upgrade FW, you can refer to the link: https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/tree/master/preloaded_firmware. It is necessary to backup before upgrade, some people did not have backed up before, this is why we write the first important thing in the link.

The correct steps: backup FW --> update FW refer to "WM-SG-SM-42 Update Preloaded AT Command FW Application Note rev. 1.4.pdf", if you want to update FW.

If update FW without backup then meet the same above issue (you can check the above programming log), you will refer to "recover guide.pdf" to recover the board after we provided the bak file.

closed 2018-10-16 by USI-ShellyWu