USIWP1Module/USI_I-NUCLEO-LRWAN1 #13: HOW TO: Receive manual?

lukblt, 2018-10-09:
How to receive manual? (with AT+Commands ? )

USI-ShellyWu, 2018-10-10:
Please refer to "4.5 Start Continue-RX for RF tests" on page 66 of AT Command Reference Manual version2.3.

USI-ShellyWu, 2018-10-10:
For two LoRa Nodes EVBs P2P connections, Please try below steps:
 
At Receiver,
 
# ATZ                                                           /* Reset */
# AT+DEFMODE=0                                      /* set to IDLE mode */
# AT+RF=20,868000000,7,0,1,0,8,0,0,0,0  /* set RF = 868MHz SF7/125KHz BW/CR1/CRC On/Preamble Length 8/No-IQ-Inverted/20dbm output power */
# AT+DEFMODE=5                                     /* set to RX mode */
 
At Transmitter,
 
# ATZ                                                           /* Reset */
# AT+DEFMODE=0                                      /* set to IDLE mode*/
# AT+RF=20,868000000,7,0,1,0,8,0,0,0,0  /* set RF = 868MHz SF7/125KHz BW/CR1/CRC On/Preamble Length 8/No-IQ-Inverted/20dbm output power */
# AT+DEFMODE=4                                     /* set to TX mode */
 
# AT+TXT=1,48454C4C4F                          /* transmit 1 packet contains raw data 'HELLO'  */
 
At RX Event,
+RCV=-1,5,48454C4C4F                           /* receive 1 packet contains raw data 'HELLO'  */

lukblt, 2018-10-10:
there is a error  in "4.5 Start Continue-RX for RF tests" on page 66 of AT Command Reference Manual version2.3

# ATZ /* SYSTEM RESET */
# AT+DEFMODE=0 /* Set Module in Idle Mode (Exit LoraWAN Mode) */
/* SET RF = 868MHz SF7/125KHz BW/CR1/CRC On/Preamble Length 8/No-IQ-Inverted/20dbm output power */
# AT+RF=20,868000000,7,0,1,1,8,0
# AT+DEFMODE=3 /* START CONTINUE-RX MODE */

Please correct DEFMODE=3 to DEFMODE=5

USI-ShellyWu, 2018-10-10:
AT+DEFMODE=3  Start Continue-RX test for sensitivity test
AT+DEFMODE=5  Set listen mode
This is different work mode.

closed 2018-10-16 by USI-ShellyWu