USIWP1Module/USI_I-NUCLEO-LRWAN1 #16: Can't update the firmware

mvladic, 2019-01-23:
I had similar situation like the one described here:

https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/issues/9

Here is my log:

```
>sm42_update.bat 9 wm-sg-sm-42_firmware_v3.7.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 2.871002
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 3438343360369009
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!
```
So, my unique device id is **3438343360369009**. Could you send me .BAK file to martin@envox.hr. Thanks!

USI-ShellyWu, 2019-02-11:
Please refer to https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20recover%20guide.pdf
And provide your "device ID".

pevandenburie, 2019-03-05:
Hi everyone,
I have the same issue: I understand how to recover using ```sm42_update.bat```, however how can I generate a .BAK when the ```sm42_backup.bat``` fails on "Error: sm42 interface seems abnormal" ?
(Lucky me, I have the Device ID)
This point is not described in the "recover guide.pdf"...
Thanks!

USI-ShellyWu, 2019-03-06:
Please kindly send the Device ID, then we will provide the bak file later.

pevandenburie, 2019-03-06:
Many thanks!
I have a bunch of them:

    34-38-34-33-59-36-91-09
    00-38-A5-66-46-36-4F-22
    34-38-34-33-73-36-95-09
    34-38-34-33-5C-36-99-09
    E2-4F-43-FF-FE-44-C4-6C

Actually, is there a tool you could share to generate these ```.bak``` ?

Or another solution could be to provide the original firware in ```.bin``` format, so I could simply drap-and-drop into the board...

Thanks!

USI-ShellyWu, 2019-03-08:
Please send your mail then the bak files will be sent to you.

pevandenburie, 2019-03-08:
My address: pevandenburie (at) yahoo.fr
Thanks!

pevandenburie, 2019-03-12:
Hi ShellyWu,
By chance, did you have the time to send me the .bak file? I did not received...
Thanks!

USI-ShellyWu, 2019-03-14:
Hi, sorry that reply late because of my vacation. The bak files have been sent to your mail just now, please kindly check them. If OK, I will close this issue.

closed 2019-03-19 by USI-ShellyWu