USIWP1Module/USI_I-NUCLEO-LRWAN1 #17: Unexpected current draw!

pbecchi, 2019-01-25:
Running the module in deep slep mode a find a 750uA current rather than 150uA i should expect.
Running the module at 4.2 mHz i get 1.2mA where it should be 0.6 mA.

There is a constant 600uA current lost inside the module!!

Any way to avoid that?  In a ultra low power device this is a very high current!

pbecchi, 2019-01-30:
I just want to add some info
The current was measured at the 3v board input pin.
Measuring thr current  at the MCU jumper (with 0v restor removed) i get 150uA  as expected.
This mean that the board draw about 600 uA but i dont know why!

hynci117, 2019-02-09:
Similar problem here. With the firmware version 3.7 the sleep current is cca 80 uA higher than with previous veriosn of modules. Makes it **unusable for battery operated devices**.

USI-ShellyWu, 2019-02-11:
Can we know the commands both of you entered for sleep mode and deep sleep mode?

hynci117, 2019-02-11:
command: 

> AT+PS

response: 

> AT+PS=1,1,0,5000                                                              
> OK                                                                           
> \# 

command: 

> AT+SLEEP

response:

> OK                                                                        
> \# +PS=1,1

pbecchi, 2019-02-11:
Just to be more precise: i am programming your board using c++ and setting sleep and stop mode.
What i have noted is that measuring current at the 3v board input i get .8 mA while measuring using the power jumpers inside the board i get in stop mode 5uA on the Stm32 and less than 10uA on LoRa chip.
This means that there is a high current drawn by the board not by the module.

hynci117, 2019-02-12:
Oh, then this is a different issue. I will better open a new one.

USI-ShellyWu, 2019-02-13:
@pbecchi As understood, you measured the nucleo board is 7mA when LoRa module is in deep sleep mode. The measured point is total 3.0V input for the board, right?
when LoRa module is in stop mode, your measured the STM32 is 5uA and LoRa chip is less than 10uA. Could you please tell which test point is? The shield board schematic can be refer to https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/tree/master/Schematics.
On the nucleo shield board there have one LoRa module (it contains ST MCU and LoRa chip 1272), and three sensor chips. Supposed the power of LoRa chip 1272 inside could not be measured.

pbecchi, 2019-02-13:
Thanks.
I have used RF &MCU power consumption measurement jumpers JP14 and JP15 after removal of 0 ohm resistors.

USI-ShellyWu, 2019-02-13:
Did you remove JP1, JP3 and JP4?

pbecchi, 2019-02-13:
Yes fully removed.

USI-ShellyWu, 2019-02-13:
Sorry for confused me.
Question 1:  7mA is total current of the nucleo board when lora is in deep sleep mode. JP1, JP3 and JP4 all are removed, right?
Question 2: 7mA is measured in deep sleep mode, 5uA and 10uA are measured in stop mode. They are different working mode.
Question3: After removed JP1, JP3 and JP4, the three sensor is not powered on, but I2C data/clock are pulled high by 3V3, there should have some leakage current to the three sensor chipsets. Please check it.

pbecchi, 2019-02-13:
Answer to your questions:
1 and 2 : at 3v input i measure 0.78mA or 780uA for sleep mode and 670uA for stop mode.
3 i understand this could be the reason, but how can i check leak current without removing pullup resistors?

USI-ShellyWu, 2019-02-13:
Question3: It is better to remove the pulled up resisitors on I2C data/clock. If not remove the resistors, it is suggested to not remove JP1, JP3 and JP4, then meausre the currents one by one: JP1, JP3 and JP4 for sensors, JP14 and JP15 for Lora module with removing SB36 and SB37.

pbecchi, 2019-02-13:
Thanks i am happy with jumper measurement since my Final design will be a custom board with  module Wg sg sm 42 .
I have build a prototype with some error on power supply connections: VDDA , and VDD Rf are not connected. The module works but show an abnormal draw of about 15mA.
Can this be due to missing power connections?

USI-ShellyWu, 2019-02-14:
We do not measure the power when VDD RF are not connected. It is recommended to refer the schematic of nucleo shield board and connect all power pins.

closed 2019-02-28 by USI-ShellyWu