USIWP1Module/USI_I-NUCLEO-LRWAN1 #18: Module hangs after going to LoRa mode after sleep

hynci117, 2019-02-09:
The issue is as follows:
- after reset issue the AT+SLEEP command, response (the default mcu sleep mode is set to 1 - stop mode)
> # OK                                                                        
> # +PS=1,1
- wake modue with at command, response:
> +PS=0,1                                                                       
> OK                                                                           
> # 
- module is now in idle, set it to LoRa mode with AT+DEFMODE=6, response:
>  OK                                                                         
> #  
Now the module does not respond to any command.
Am I missing something? Is there a workaround?

USI-ShellyWu, 2019-02-11:
Can we know your test FW version?

hynci117, 2019-02-11:
The FW is 3.7

hynci117, 2019-02-26:
This sequence during init (setting default sleep and persisting in eeprom) causes the module to stop responding:

AT+VER
\# +VER=1.0.1,3.7,SM42                                                         
OK
                                                                         
AT+DEFMODE=6
\# OK
                                                                    
AT+PS=0,1
\# OK
                                                                       
AT+WDCT
\# OK
                                                                          
AT+DEFMODE=0
\# OK
  
From now the module does not respond to any command and needs to be reset by external pin.

USI-ShellyWu, 2019-02-27:
I have tried as below commands
# AT+PS1,1,0,5000
OK
# AT+SLEEP+PS=1,1
 
"Press Enter on my keyboard to wake up SM42"
+PS=0,1
OK
# AT+DEFMODE=6OK
# AT+VER1.0.1,3.7,SM42
OK
# AT+BAND0
OK
# AT+PS1,1,0,5000
OK
# AT+ADDR11,22,33,44
OK
# AT+VER1.0.1,3.7,SM42
OK

After setting AT+DEFMODE=6, the module can still respond some AT commands like "AT+VER", "AT+ADDR" "AT+BAND". 
I will try your commands later on.

USI-ShellyWu, 2019-02-28:
I tried for this twice, not found the module hang up.
The first time===================
# AT+PS  1,1,0,5000
OK
# AT_SLEERROR_UNKNOW_COMMAND
# AT+SLEEP  +PS=1,1
+PS=0,1
OK
# AT+VER  1.0.1,3.7,SM42
OK
# AT+DEFMODE=6  OK
# AT+VER  1.0.1,3.7,SM42
OK
# AT+BAND0
OK
# AT+DEFMODE=0  OK
# AT+VER  1.0.1,3.7,SM42
OK
# AT+DEFMODE  0
OK

the second time=================
# AT+PS  1,1,0,5000
OK
# AT+SLEEP  +PS=1,1
+PS=0,1
OK
# TERROR_UNKNOW_COMMAND
# AT+VER  1.0.1,3.7,SM42
OK
# AT+DEFMODE=6  OK
# AT+PS=0,1  OK
# AT+WDCT=0  OK
# AT+DEFMODE=0  OK
# AT+VER   1.0.1,3.7,SM42
OK
# AT+BAND0
OK

USI-ShellyWu, 2019-02-28:
Above some error message is because key in the wrong commands, not impact the result.

hynci117, 2019-02-28:
Thanks for the update. The sequence you entered works for me too. But there is a difference, you sent AT+WDCT=0 which resets the eeprom insted of AT+WDCT which stores the current settings. 
The shortest sequence of  commands to reproduce after module reset:
AT+PS=0,1\r
AT+WDCT\r
AT+SLEEP\r
AT\r (just to wake up)
AT+DEFMODE=0\r
no response after

USI-ShellyWu, 2019-03-01:
# AT+VER  1.0.1,3.7,SM42
OK
# AT+PS=0,1 OK
# AT+WDCT OK
# AT+VER 1.0.1,3.7,SM42
OK
# AT+SLEEP +PS=1,1
+PS=0,1
OK
# AT+DEFMODE=0 OK
# AT+VER 1.0.1,3.7,SM42
OK
# AT+BAND 0
OK
# AT+DEFMODE=6 OK
# AT+VER 1.0.1,3.7,SM42
OK
# AT+ADDR 00,80,8c,3c
OK
# AT+VER 1.0.1,3.7,SM42
OK
# AT+BAND 0
OK
# 

Re-tried your commands sequence, the hang up issue did not happen.

hynci117, 2019-03-01:
I am getting the problem consistently. I tried on two other modules and one is working okay, the second one hangs in a different way (not reaching the Tx Done). Can it be related to the state of the wake-up pin or other wiring problems related to soldering? (One of the faulty modules is on the original Nucleo shield)
If not, I will regard the modules as faulty and we can close this.

USI-ShellyWu, 2019-03-04:
I have tried this commands sequence on one Nucleo shield board without any change on the board, because there has only one shield board on hands. 
After enter sleep mode with "AT+SLEEP", I used with pressing "Enter" in my keyboard that input into the module through UART, is it same?
Then I tried to set working mode.

hynci117, 2019-03-04:
Yes, that should be the same. It works for you?

USI-ShellyWu, 2019-03-05:
The module still works after above commands sequence, not happened with any hang up.

hynci117, 2019-03-05:
Okay, lets asume it is a hardware related problem and close this. Thank you for helping me.

closed 2019-03-05 by hynci117