USIWP1Module/USI_I-NUCLEO-LRWAN1 #19: High current in sleep mode for fw 3.7

hynci117, 2019-02-12:
command: 

> AT+PS

response: 

> AT+PS=1,1,0,5000                                                              
> OK                                                                           
> \# 

command: 

> AT+SLEEP

response:

> OK                                                                        
> \# +PS=1,1

_Originally posted by @hynci117 in https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/issues/17#issuecomment-462271357_

USI-ShellyWu, 2019-02-12:
Under checking.

USI-ShellyWu, 2019-02-14:
Please measure the power with FW3.3, for FW3.7 we are checking the reason.

hynci117, 2019-02-26:
Update?
I was able to compare the 2.9 firmware with the 3.7 on the same board, with the same commands. The measured currents are as follows:
AT+PS=0,2: current consumption fw2.9 = 94uA, fw3.7 = 94uA (both the same)
AT+PS=0,1: current consumption fw2.9 = 6uA, fw3.7 = 94uA (3.7 much higher)

USI-ShellyWu, 2019-02-27:
Yes, compared the currents with FW3.3 and 3.7, it is higher with FW3.7. We are trying to fix the problem in next FW.