USIWP1Module/USI_I-NUCLEO-LRWAN1 #20: patch for v1.2.1

pevandenburie, 2019-02-21:
Hi there,

I'm currently modifying the sm42-lrwan_v1.1.5.patch to adapt to v1.2.1 : changing files path, applying dos2unix, etc...

Unfortunately, the folders structure changed a lot, and moved from:

    "Projects/Multi/Applications"
to

    "Projects/{B-L072Z-LRWAN1, STM32LxxxRx-Nucleo}/Applications"

I understood AT_SLAVE is only for B-L072Z-LRWAN, but some files like "stm32l0xx_hw_conf.h" 

    Projects/Multi/Applications/LoRa/AT_Slave/inc/stm32l0xx_hw_conf.h

is not anymore in "Projects/B-L072Z-LRWAN1":

```
find -name stm32l0xx_hw_conf.h
./Projects/STM32L053R8-Nucleo/Applications/LoRa/End_Node/Core/inc/stm32l0xx_hw_conf.h
./Projects/STM32L053R8-Nucleo/Applications/LoRa/PingPong/Core/inc/stm32l0xx_hw_conf.h
./Projects/STM32L053R8-Nucleo/Applications/LoRa/AT_Master/Core/inc/stm32l0xx_hw_conf.h
./Projects/STM32L073RZ-Nucleo/Applications/LoRa/End_Node/Core/inc/stm32l0xx_hw_conf.h
./Projects/STM32L073RZ-Nucleo/Applications/LoRa/PingPong/Core/inc/stm32l0xx_hw_conf.h
./Projects/STM32L073RZ-Nucleo/Applications/LoRa/AT_Master/Core/inc/stm32l0xx_hw_conf.h
```

So I really don't know how to fix the patch here.
Any clue?

USI-ShellyWu, 2019-02-27:
The new patch for ST Irwan1.2.1 will be released soon.

pevandenburie, 2019-03-07:
Ok great! Let us know ;-)
Meanwhile is there some source code to share, even for a previous version? Indeed the ```en.i-cube_lrwan``` version 1.1.0 (corresponding to the patchs) is not available anymore for download.

AnTrii, 2019-09-24:
There is ST I-CUBE-LRWAN 1.2.2 already, and it still doesn't support USI® I-NUCLEO-LRWAN1. And I cannot find the 1.1.5 sources anywhere. Anyone knows where to find I-CUBE-LRWAN compatible with I-NUCLEO-LRWAN1? Thank you in advance.

USI-ShellyWu, 2019-09-26:
Patch for I-CUBE-LRWAN 1.2.2 can be downloaded from https://github.com/USIAndyrich/USI_I-NUCLEO-LRWAN1/tree/master/I-CUBE-LRWAN_patchs/I-CUBE-LRWAN%201.2.2 
Please notice that AT-Slave application is removed in ST software kit.

AnTrii, 2019-09-27:
Thank you very much!

AnTrii, 2019-10-31:
@pevandenburie, Write a comment on results, In case of negative, here is a link to archive with both ST I-CUBE-LRWAN v.1.2.2 and USI patch for it: [https://app.box.com/s/a3pvl9ptdhvs2c0ubfs60vb35vu15mkf](url)

USI-ShellyWu, 2019-11-01:
The patch for 1.2.2 can work in I-CUBE-LRWAN 1.2.2 and 1.2.3 both. Downlink: https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/tree/master/I-CUBE-LRWAN_patchs/I-CUBE-LRWAN%201.2.2