USIWP1Module/USI_I-NUCLEO-LRWAN1 #22: Configuration issues using AT commands with wm-sg-sm-42_firmware_v3.7.hex

pevandenburie, 2019-03-19:
Hello
I successfully updated my I-NUCLEO-LRWAN1 boards to ```wm-sg-sm-42_firmware_v3.7.hex``` using ```sm42_fw_update_tool_v1.1-1```.

It seems to work and I have the AT commands. However, I cannot change the BAND to 0 (EU-868 Band), and stays fixed to 1 (US-Band):
    
    # AT+VER1.0.1,3.7,SM42
    OK
    # AT+BAND=0OK
    # AT+BAND11
    OK
    # AT+BAND1
    OK

Additionally, I cannot change Data Rate nor Data Rate of Rx2 window to 5:

    # AT+DR=5ERROR_INVALID_ARGUMENTS
    # AT+DR0
    OK
    # AT+RX2DR=5ERROR_INVALID_ARGUMENTS
    # AT+RX2DR3
    OK

Is used to work, what did I do wrong?

Thanks!

USI-ShellyWu, 2019-03-20:
Please try below
AT+BAND=0
AT+WDCT
ATZ

pevandenburie, 2019-03-25:
That worked! Good to know...
Thanks!

closed 2019-03-25 by pevandenburie