USIWP1Module/USI_I-NUCLEO-LRWAN1 #23: Problem with end-node application!

VelcoH, 2019-06-14:
Hi, so I have tried running the end-node application in v.1.1.5, and v.1.2.1. 

I'm using:
- Nucelo STM32L053 
- I-NUCELO-LRWAN

I have the same issue as this: https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/issues/8

My application:
- Connecting LoRa end node to a LoRa gateway, and sending sensor data.

Once I flash the software onto the nucelo, the only thing that pops up in UART is 
OTAA
DevEui
AppEui
AppKey

And nothing else.

USI-ShellyWu, 2019-06-17:
1. Connecting LoRa end node to a LoRa gateway, and sending sensor data. --> Did the LoRa gatewady get the sensor data successfully?
2.  Once I flash the software onto the nucelo,  --> how to flash the software and which version is the software?
3. Please provide the detailed:
OTAA
DevEui
AppEUI
APPKEY

VelcoH, 2019-06-19:
Thanks for replying, @USI-ShellyWu.

Here are the answers to your questions:

1. No, unfortunately, I was unable to connect to the LoRa gateway (look below).

2. So, without using the patch. I used the version 1.1.5 and the project located at: "STM32CubeExpansion_LRWAN_V1.1.5\Projects\Multi\Applications\LoRa\End_Node\SW4STM32\STM32L053R8-Nucleo\sx1272mb2das". Furthermore, I flashed the firmware onto the nucelo, without patching anything. 

3. The UART is shown below. After all this, nothing further happens.

OTAA
DevEui= 39-38-34-32-71-37-91-09
AppEui= 70-B3-D5-7E-D0-01-D8-81
AppKey= FB CB 25 42 B7 39 68 49 CB 91 7E A6 1B 30 1F E1
VERSION: 44101150

My questions to you are:
1. Do you need to use the patch to run the end node application?
2. After I have applied the patch (v 1.1.5), do I flash the Nucelo board or USI extension board directly?
3. The path "End_Node\SW4STM32\STM32L053R8-Nucleo\" has three different folders; "sx1272mb2das", "sx1276mb1las" and "sx1276mb1mas". Which one do I use?
4. Can you provide me steps to run an end node application?

USI-ShellyWu, 2019-06-20:
Yes, it is necessary to use the patch.
For end node application, please check in the link: https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/I-CUBE-LRWAN_patchs/I-CUBE-LRWAN%201.1.5/README.md