USIWP1Module/USI_I-NUCLEO-LRWAN1 #25: How to change the power level of Transmission abd Reception

sriniash, 2019-12-03:
What AT commands are to be used to change the power level and then get back to the application to communicate with the gateway at a much reduced power?
The default seems to be set at 20dBm always based on the current being measured at the VCC_RF.
Also is there any command to check the RF settings? e.g. AT+RF should it give back any parameters?

USI-ShellyWu, 2019-12-18:
AT+RF is used to set power level/bw/frequency and other parameters. If the return value is OK, the RF setting should be all right. You can use the power meter to measure on the module's RF. At most time AT+RF is used to test RF for verification or getting certifications. May I know the reason that you want to change power level.

sriniash, 2019-12-18:
**My requirement is to change the power levels in the actual application and gauge the power and range with reduced power**

I am able to run the AT commands when connected to the Nucleo board through the STM32 ST LNK utility. I can change the power etc. when in this mode using the AT commands, but the moment I go back to the main application by disconnecting from the Nucleo board, then the  values I had entered are no longer valid even if I saved to EEPROM. It now starts up with full power even if I had reduced the power using the AT commands.

USI-ShellyWu, 2019-12-19:
When in the actual application, now it can not support to set the power level with AT commands.

sriniash, 2019-12-19:
So I would have to make these changes in the main application? Can you point me to where to make the changes?

USI-ShellyWu, 2019-12-19:
May we know the reason you would like to set the power level in the application. As we learned, it is not necessary to set the power level in LoRaWan application.

sriniash, 2019-12-19:
To optimize the power taken by the RF in battery powered applications where the full range is not required.

USI-ShellyWu, 2019-12-20:
The power setting is not active in the current FW that follow all the power spec in LoRaWan,