USIWP1Module/USI_I-NUCLEO-LRWAN1 #26: Porting from fw 3.7 to 4.0.1

hynci117, 2020-01-20:
Hello, we received a batch of modules with the new firmware (ver. 4.0.1). Our current application used version 3.7 modules and it is not working with the updated firmware. Checking the changelog does not reveal any reason for this problem. Could you update the documentation please?
The current behavior:
AT+VER
OK
 # +VER=1.0.1,4.0.1,SM42,1.1                                                                        
AT+DEFMODE=6
# ERROR_NOT_SUPPORTED

slc0508, 2020-02-03:
Can you provide any error message or defective symptom for our investigation?

hynci117, 2020-02-03:
One symptom is that the tx test is not working (page 65 in AT command manual, version 2.5 - 1.1 Start Continue-TX for RF tests). However other examples do not work as well. I was able to find out that the +DEFMODE command is not working as expected (in previous versions is ok). It was shown in my first post:

\# AT+DEFMODE=0        // <= set the module to idle mode
OK
\# AT+DEFMODE
6    // <= this should return "0" - idle mode. instead the module stays in the LoRaWAN mode
OK

lugiaking, 2020-02-03:
Since AT+DEFMODE 0 ~ 5 is used for testing purposes, it is a kind of temporarily used, so you can set AT+DEFMODE from 0~5 directly for test,and then reset module by ATZ for return to mode 6, so we removed mode 6 for AT+DEFMODE and only can query the value stored in eeprom (because AT+WDCT and ATZ was not performed), so can you please kindly skip the query steps? the following command sequence for TX test is for your reference, thank you so much !!

@ Start Continue-TX 
# ATZ                           /* SYSTEM RESET */ 
# AT+RF=20,868000000,7,0,1   /* SET RF@868MHz with SF7/125KHz BW/CR1/20dbm OUTPUT PWR */ 
# AT+DEFMODE=2         /* START CONTINUE-TX */
# ATZ                              /* STOP CONTINUE-TX by RESET command */

 @ Generate TX Packet 
 # ATZ                      /* SYSTEM RESET */ 
# AT+RF=20,868000000,7,0,1   /* SET RF@868MHz with SF7/125KHz BW/CR1/20dbm OUTPUT PWR */ 
# AT+DEFMODE=4            /* START PACKET TRANSMISSION MODE */ 
# AT+TXT=100,48454C4C4F     /* Generate 100 PACKETS with PAYLOAD ‘HELLO’ */ 
+TX: Done                  /* Waiting the TX done event, it indicates all packets was generated */
# ATZ                      /* or STOP TX by RESET command */

hynci117, 2020-02-03:
Thanks, that explains some of the problems. But I tried the command sequences you suggested on two different modules and no tx signal was detected. Can you please confirm that this sequence works on 4.0.1 modules? In that case we might have a hardware issue.

lugiaking, 2020-02-04:
Hello, I have confirmed this two command sequences work on my I-NUCLEO-LRWAN1 board using a spectrum analyzer, if you are not sure the hardware is ok or not, you can try it on the EVB or  I-NUCLEO-LRWAN1 board first, thanks!!

hynci117, 2020-02-06:
Hello and thanks. This helped. But in the example you gave it is needed to change order of the commands, otherwise the rf settings command has no effect:
ATZ
AT+RF=20,868500000,7,0,1   // set 868.5 MHz
AT+DEFMODE=4
AT+TXT=100,48454C4C4F  // transmits on 868.0 MHz

When changing rf settings AFTER changing mode, it works okay:
ATZ
AT+DEFMODE=4
AT+RF=20,868500000,7,0,1   // set 868.5 MHz
AT+TXT=100,48454C4C4F  // transmits correctly on 868.5 MHz

For me this is issue ok. But it would be nice if you could update the documentation to match the current version of the AT commands (removed mode 6, when can the AT+RF command be issued and also I noticed that wake on uart does no longer work, maybe there are other differences?).
Thank you for support.

lugiaking, 2020-02-07:
Hello, yes the TX signal will always at 868MHz if command sequence is AT+RF then AT+DEFMODE, thank you for update, I will update the document for you soon, and regarding the wake on uart function, currently, the default MCU sleep mode is 2 (standby mode) and only can be waked by module pin-11 (SYS_WKUP1), so if you want to wake module by uart, you may change the sleep mode to mode 1 (stop mode) by the command below, but mode 1 (stop mode) will have more power consumption a little bit than mode 2 (standby mode)!!

AT+PS=0,1     <-- using stop mode as system sleep mode 
AT+WDCT      <-- write changes to eeprom

sincerely thanks!

lugiaking, 2020-02-07:
Hello, Sorry I forgot a step (ATZ) for reboot in my previous commented, the new sleep mode will no effect until module rebooted. 

reboot command:
ATZ