USIWP1Module/USI_I-NUCLEO-LRWAN1 #29: USI_I-NUCLEO-LRWAN1 Does not answer to join accept

JulienDevillers, 2020-02-12:
Hello, 
I run the at_master example from I-NUCLEO-LRWAN2 eval pack.

I did update the firmaware to set  my app id and app key using LoRa_SetAppID()  and LoRa_SetKey().
It worked correctly. I'm collecting data through TTN.

Then I did reboot the device and from that point, the app is no more functionnal.
Using gateway logs from TTN. I can see that after a join request, the gateway sent a "Join Accept" frame.  (see screenshot)

[image]

If I sniff the serial port for AT commands,  I can see that the modem answers "OK" to the "at+join=1" but does not send the "+JoinAccepted" as I think it should. See logs below :

`

Time [s] |   | To modem | From modem |   |  
-- | -- | -- | -- | -- | --
4.219882000000000 |   |   | '0' |   |  
4.676934000000000 |   |   | \r |   |  
4.677020500000000 |   |   | # |   |  
4.677108000000000 |   |   | ' ' |   |  
5.377679000000000 |   | A |   |   |  
5.377766000000000 |   | T |   |   |  
5.377853000000000 |   | E |   |   |  
5.377940000000000 |   | = |   |   |  
5.378027000000000 |   | 0 |   |   |  
5.378114000000000 |   | \r |   |   |  
5.378921500000000 |   |   | O |   |  
5.379008500000000 |   |   | K |   |  
5.379095500000000 |   |   | \r |   |  
5.379183000000000 |   |   | \n |   |  
5.379343000000000 |   |   | \r |   |  
5.379430000000000 |   |   | # |   |  
5.379517000000000 |   |   | ' ' |   |  
7.383442500000000 |   | A |   |   |  
7.383529500000000 |   | T |   |   |  
7.383616500000000 |   | + |   |   |  
7.383703500000000 |   | A |   |   |  
7.383790500000000 |   | P |   |   |  
7.383877500000000 |   | P |   |   |  
7.383964500000000 |   | E |   |   |  
7.384052000000000 |   | U |   |   |  
7.384139000000000 |   | I |   |   |  
7.384226000000000 |   | = |   |   |  
7.384313000000000 |   | 7 |   |   |  
7.384400000000000 |   | 0 |   |   |  
7.384487000000000 |   | b |   |   |  
7.384574000000000 |   | 3 |   |   |  
7.384661000000000 |   | d |   |   |  
7.384748000000000 |   | 5 |   |   |  
7.384835000000000 |   | 7 |   |   |  
7.384922000000000 |   | e |   |   |  
7.385009000000000 |   | d |   |   |  
7.385096000000000 |   | 0 |   |   |  
7.385183000000000 |   | 0 |   |   |  
7.385270000000000 |   | 2 |   |   |  
7.385357000000000 |   | a |   |   |  
7.385444000000000 |   | 5 |   |   |  
7.385531000000000 |   | 2 |   |   |  
7.385618000000000 |   | f |   |   |  
7.385705000000000 |   | \r |   |   |  
7.387433500000000 |   |   | O |   |  
7.387520500000000 |   |   | K |   |  
7.387607500000000 |   |   | \r |   |  
7.387695000000000 |   |   | \n |   |  
7.387855000000000 |   |   | \r |   |  
7.387942000000000 |   |   | # |   |  
7.388029000000000 |   |   | ' ' |   |  
8.389777000000000 |   | A |   |   |  
8.389863500000001 |   | T |   |   |  
[... 194 lines cut ...]
30.510670999999999 |   |   | K |   |  
30.510757999999999 |   |   | \r |   |  
30.510845499999999 |   |   | \n |   |  
30.511005999999998 |   |   | \r |   |  
30.511092500000000 |   |   | # |   |  
30.511180000000000 |   |   | ' ' |   |  
32.515253999999999 |   | A |   |   |  
32.515340999999999 |   | T |   |   |  
32.515428000000000 |   | + |   |   |  
32.515515000000001 |   | A |   |   |  
32.515602000000001 |   | P |   |   |  
32.515689000000002 |   | P |   |   |  
32.515776000000002 |   | E |   |   |  
32.515863000000003 |   | U |   |   |  
32.515949999999997 |   | I |   |   |  
32.516036999999997 |   | = |   |   |  
32.516123999999998 |   | 7 |   |   |  
32.516210999999998 |   | 0 |   |   |  

`

Do you have any idea of what may happen ?

Thanks
Julien

lugiaking, 2020-02-25:
Hello,

I am sorry for the late reply, regarding to the screen-shot, I can see data rate for the join-accept frame from the server is DR0 on RX2, but I am not sure if the default data rate for RX2 is DR0 or DR3, you can try to change the data rate for RX2 by the command below: 

AT+RX2DR=3

I want to know the fw version you are using for the test, can you please share me the fw version by the command below?  thanks!!

AT+VER

JulienDevillers, 2020-02-29:
Hello,

After 2 weeks with everything unplugged. I did power on the gateway and the node.
It worked as it should. Then  I reboot the node and immediatly get the above behaviour.
I don't if this may help.

AT+RX2DR=3 did not change anything.

FW Version is 1.0.1 4.0.1 SM42 1.1

szumbrunn, 2020-03-02:
Hi guys
I've seen this issue too. For example with Swisscom LPN using the latest firmwares 4.0.1 and 4.0.2, the default setting of the receive datarate RX2DR changed from 0 to 3. So make sure you set this correctly before sending. 
Also: There is currently a bug in the firmware (4.0.x) that it does not save the RX2DR with WDCT sometimes. I have not found a solution yet. Hopefully, they will provide us with a fix very soon, we have a production release awaiting and got modules with the new buggy firmware.

MilorES, 2020-04-27:
Couldn't we use a version that didn't have the problem? (looks like it's from 4.0.x)

reinaldoaf, 2020-06-30:
I have same problem, when I try connect to at+join=1, the shield don´t stop to answare.. please have any to do? the problem is only with OTAA, ADB Mode work?
[image]
[image]

lugiaking, 2020-07-22:
Hello 

yes, the module would not stop to transmit join request until succeed to join on the concentrator,  and after firmware version 4.0.2 the module will report join status proactively.  thanks!!

Jezek-David, 2021-02-08:
I have the same problem. Did you find any solutions?

reinaldoaf, 2021-02-11:
> Links:

unfortunately not, and worse, no support from the manufacturer, that is, 
it is an expensive and useless kit for me so far

I keep looking, if I find something to inform you, I would be very 
grateful if I could do the same

JulienDevillers, 2021-02-11:
Near the same here, solution was to use something else....