USIWP1Module/USI_I-NUCLEO-LRWAN1 #30: Upgrading Onboard Firmware v2.9 to v4.0.1

adamwilsonCPI, 2020-02-20:
Hi,

I've sent several emails over the last week to USI's customer service team, but got no response.

I'm evaluating 10 of your I-NUCLEO-LRWAN1 (WM-SG-SM-42) boards for use in an R&D project.

I've noticed that they work for up half an hour whilst sending/receiving data, then the modem will lock up and become unresponsive on either a JOIN (via OTAA) or SEND command. I checked out your firmware changelog for this reloaded firmware, and noticed that this is a known issue that was fixed in v3.5.
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/CHANGELOG.txt

All 10 were bought from Farnell UK and have firmware version "1.0.1, 2.9" preinstalled. 

I've therefore been trying to install the latest firmware version (v4.0.1) using an STLINKv2 programmer, which successfully connects to the STM32, and your SM42 firmware update tool (v1.1):
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/tree/master/preloaded_firmware/v4.x/wm-sg-sm-42%20firmware%20v4.0.1/image%20for%20sm42_fw_update_tool
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/sm42_fw_update_tool_v1.1-1.zip
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20Update%20Preloaded%20AT%20Command%20FW%20Application%20Note%20rev.%201.4.pdf

However, I get the following terminal output:
"C:\Users\adam.wilson\Downloads\sm42_fw_update_tool_v1.1-1\bin>sm42_update.bat 6 sm42_monolithix_v4.0.1.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v36 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 2.910236
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 303634327C376602
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!"

When I try to perform a backup before updating the firmware, I get the following error:
"C:\Users\adam.wilson\Downloads\sm42_fw_update_tool_v1.1-1\bin>sm42_backup.bat 6
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ..."

I've been reading the issues on the module's GitHub page, which suggest that this is a common issue and that the only solution is for USI to provide a backup file (.bak) for each specific device ID. 

Can you provide a .bak for device ID "30-36-34-32-7C-37-66-02", so I can test that this works?
If it does, I'll also need backup files for the other 9 modems.

If not, I'd appreciate some advice on how to get the latest version working on these modules.

Failing that, is there anywhere that I can purchase hardware with the latest firmware installed?

Many thanks,

Adam

Serdrutsa, 2020-02-21:
Hi,
I have a similar problem. Is it possible to get the bak file for the following ID?
   Info : found unique device id = 3036343256377403

Thanks,
Serghei

lugiaking, 2020-02-25:
Hello,  I uploaded two 7z files for id '3036343256377403' and '303634327C376602', please try it again, thank you!

lugiaking, 2020-02-25:
please let me know if you cannot find the 7z files, thanks!

lugiaking, 2020-02-25:
Sorry, if you get the 7z files, please unzip them before use, thank you!

adamwilsonCPI, 2020-02-25:
> Hello, I uploaded two 7z files for id '3036343256377403' and '303634327C376602', please try it again, thank you!

Thank you :-)

I'm afraid I can't find the compressed file, where did you upload them to?

Serdrutsa, 2020-02-25:
Yep.. A hint where to look for the 7z files could help.. :)

lugiaking, 2020-02-26:
Sorry, I guess 7z is unrecognized on the chat, so I re-uploaded them in zip format. 
please try it again, thanks!!

[3036343256377403.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4255263/3036343256377403.zip)
[303634327C376602.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4255266/303634327C376602.zip)

adamwilsonCPI, 2020-02-26:
> Sorry, I guess 7z is unrecognized on the chat, so I re-uploaded them in zip format.
> please try it again, thanks!! [...]

Hi again,

This has worked for me, thank you :-)

I followed the recovery guide available here:
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20recover%20guide.pdf

When 
_"NOTE!! nvm must be restored ...
DO NOT TERMINATE the process or the window.
to ensure target is in running state, please unplug SWD wire and REPOWER target

then please hit ENTER key while ready to continue ...."_
came up, I powered down the FTDI cable, LoRa shield, and my ST-LINKv2, then powered them all on again, then pressed enter.

Is it easy to modify these backup files for a different device ID? Is it just the filename that needs changing? I now have another ten shields to update, each with firmware v2.9 and different device IDs, and I need to be self-reliant at generating these backup files if I'm going to use them.

Many thanks,

Adam

Serdrutsa, 2020-02-26:
Many thanks!
The bak file worked for me also.

Can I ask you just one more question?
Why the DEFMODE is not saved? I set any other value (0-5) and it always reads back 6 (see picture).
[image]
The same effect was with older FW version and with latest v.4.0.2.

lugiaking, 2020-02-27:
sorry to let you inconvenience, yes, AT+DEFMODE always will report '6' from fw v4.x. we will fix it in next release.

adamwilsonCPI, 2020-02-27:
> sorry to let you inconvenience, yes, AT+DEFMODE always will report '6' from fw v4.x. we will fix it in next release.

Is it easy to modify these backup files for a different device ID? Is it just the filename that needs changing? I now have another ten shields to update, each with firmware v2.9 and different device IDs, and I need to be self-reliant at generating these backup files if I'm going to use them.

hynci117, 2020-02-27:
Similar problem here - upgrading a module failed in the middle and it will not upgrade or backup. Can you provide the bak for me as well? The device eui is: E24F43FFFE44C76F

sm42_update.bat 19 sm42_monolithix_v4.0.1.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : starting to backup nvm ...
................................................................................
................................................................................
................................................................................
................
Info : backup nvm is valid
Info : current fw version in module:  3.700000
Info : found eui: E24F43FFFE44C76F
Info : finish nvm backup.
Info : auto-selecting first available session transport "hla_swd". To override u
se 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results mi
ght differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.256353
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 3036343267378D03
Info : nvm was backup in '3036343267378D03.bak'
Info : stm32lx_write_rdp RDP=aa...
Info : stm32lx_write_rdp OBL_LAUNCH...
STM32Lx unlocked, wait 1 sec.
Error: jtag status contains invalid mode value - communication failure
Polling target stm32l0.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
in procedure 'reset'
in procedure 'ocd_bouncer'

adamwilsonCPI, 2020-03-03:
> Sorry, I guess 7z is unrecognized on the chat, so I re-uploaded them in zip format.
> please try it again, thanks!! [...]

Hi again,

In case it got lost in the other comments, I'd appreciate a response to my previous question:

"Is it easy to modify these backup files for a different device ID? Is it just the filename that needs changing? I now have another ten shields to update, each with firmware v2.9 and different device IDs, and I need to be self-reliant at generating these backup files if I'm going to use them."

As the original poster, I'll need an answer to this before the thread can be closed. It seems that others are experiencing the same problems, too.

Many thanks,

Adam

adamwilsonCPI, 2020-03-13:
> sorry to let you inconvenience, yes, AT+DEFMODE always will report '6' from fw v4.x. we will fix it in next release.

Hi again,

I need a response from USI ASAP, as I've still got ten modules which are essentially useless because of their firmware version, and which need to be fixed. If nothing is done about it ASAP, I'm going to have to drop the USI module from this project altogether and go with another manufacturer with better support and documentation.

I'm still having problems even with firmware v4.0.2.

Regards,

Adam

adamwilsonCPI, 2020-03-23:
> sorry to let you inconvenience, yes, AT+DEFMODE always will report '6' from fw v4.x. we will fix it in next release.

I'm still waiting for assistance, please. My deadline is imminent.

If you can advise how the .bak files are produced, I should be able to sort myself out from there.

adamwilsonCPI, 2020-04-16:
I appreciate your previous advice, but you have now left me with no choice but to move to Semtech/Murata hardware instead of USI.

andrferreira, 2020-05-14:
Hi, 
I also need to upgrade the firmware of my end device and I need the bak file.
My device eui is: E24F43FFFE44D2B3

slc0508, 2020-05-20:
Hi , 
Please refer the procedure to backup the NV files one-by-one for each device

https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20Update%20Preloaded%20AT%20Command%20FW%20Application%20Note%20rev.%201.4.pdf

Follow the steps mentioned in Sec.9

lugiaking, 2020-05-20:
Hi Adam,
Sorry! I will help you to upgrade the firmware for the other ten shield-boards, so can you please help to provide device ID for the ten boards? or the bar code number on the bottom of the shield-board is also okay, if you are inconvenient to check the device id by the update tool.
BTW, since you have mentioned there are still some problems on firmware v4.0.2, so can you please also help us to understand what kind of problem you met during your developing, many thanks!

lugiaking, 2020-05-20:
Hi Andrferreira,

I uploaded the zip file for the device id  'E24F43FFFE44D2B3', please try it, thanks!!
[E24F43FFFE44D2B3.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4655348/E24F43FFFE44D2B3.zip)

lugiaking, 2020-05-20:
Hi Hynci117,

here is the zip file for the device id 'E24F43FFFE44C76F', please try it, thanks!!

[E24F43FFFE44C76F.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4655408/E24F43FFFE44C76F.zip)

adamwilsonCPI, 2020-05-20:
> Hi ,
> Please refer the procedure to backup the NV files one-by-one for each device
> 
> https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20Update%20Preloaded%20AT%20Command%20FW%20Application%20Note%20rev.%201.4.pdf
> 
> Follow the steps mentioned in Sec.9

Hi,

As discussed previously, this backup procedure does not work for firmware v2.9.

Regards,

Adam

adamwilsonCPI, 2020-05-20:
> Hi Adam,
> Sorry! I will help you to upgrade the firmware for the other ten shield-boards, so can you please help to provide device ID for the ten boards? or the bar code number on the bottom of the [...]

Hi,

Thank you for getting back to me :-)

I'll get back to you with the device IDs or barcode numbers of all of my boards ASAP, thank you.

I'll keep you updated regarding the problems I've been having while developing with v4.0.2, but chances are it could be a problem with my own firmware too, I need to do more testing.

Many thanks,

Adam

andrferreira, 2020-05-21:
Hi lugiaking,

Thanks for the file. 

How can I check the firmware version?

adamwilsonCPI, 2020-05-21:
> Hi Adam,
> Sorry! I will help you to upgrade the firmware for the other ten shield-boards, so can you please help to provide device ID for the ten boards? or the bar code number on the bottom of the [...]

Hi again,

As promised, here are the barcode IDs for each of the seven remaining boards:
#2: 88850193870217061381884
#3: 88850193870217061381978
#4: 88850193870217061380156
#5: 88850193870217061381887
#6: 88850193870217061381483
#7: 88850193870217061381689
#8: 88850193870217061382141

If you could please name the backup files to reflect the numbers I've assigned to each of these board (or the barcodes themselves), it would be appreciated.

I'd appreciate your quick response, as I have time booked in next week to perform the upgrades.

Many thanks,

Adam

lugiaking, 2020-05-21:
Hi Adam,

It needs a little of time to get the real device ID by these bar code number, so I will send you the backup files corresponding to the bar code number before next Monday. thanks!!

adamwilsonCPI, 2020-05-21:
> Hi Adam,
> 
> It needs a little of time to get the real device ID by these bar code number, so I will send you the backup files corresponding to the bar code number before next Monday. thanks!!

Hi again,

I'd greatly appreciate that, thank you :-)

Many thanks,

Adam

lugiaking, 2020-05-21:
Hello Andrferreira,

You can check the firmware version by command “AT+VER” at run time. and then you will get the versions like this '1.0.1,4.0.1,SM42,1.1', 
'1.0.1' is the version of LoRaWAN Spec, 
'4.0.1' is the firmware version, 
'SM42' is module name
'1.1' is the bootloader version

FYI.

andrferreira, 2020-05-22:
Hello @lugiaking,

The problem is the original firmware is gone and serial communication is not working. That is the reason why I asked for the .bak file. Is there any other way to check the firmware version?

Thanks for the support.

lugiaking, 2020-05-25:
Hi Andrferreira,

I got it, if firmware is gone, the flash also is blank, it indicates no firmware inside the module, so you do not need to check the firmware version,

lugiaking, 2020-05-25:
Hi Adam,

I am really sorry for the bad news, I can not get the real device ID by the bar code after check with our factory, so I still need your help to provide the update log containing the device ID like this (Info : found eui: E24F43FFFE44C76F) for each device,  and then I will send you the .bak files ASAP,  thanks!!

adamwilsonCPI, 2020-05-26:
> Hi Adam,
>
> I am really sorry for the bad news, I can not get the real device ID by the bar code after check with our factory, so I still need your help to provide the update log containing the device [...]

Hi again,

Not a problem :-) Please see unique device IDs below:
#2 - 3036343267376402
#3 - 343834335E369109
#4 - 3036343268376A02
#5 - 343834337D368E09
#6 - 3438343364369609
#7 - 343834336B369209
#8 - 3438343363369409
#9 - 3438343363368F09

Many thanks,

Adam

lugiaking, 2020-05-27:
Hi Adam,

here is the zip files for these device ID, please try it, thanks!!

[343834335E369109.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686629/343834335E369109.zip)
[343834336B369209.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686630/343834336B369209.zip)
[343834337D368E09.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686631/343834337D368E09.zip)
[3036343268376A02.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686632/3036343268376A02.zip)
[3438343363368F09.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686633/3438343363368F09.zip)
[3036343267376402.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686634/3036343267376402.zip)
[3438343363369409.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686635/3438343363369409.zip)
[3438343364369609.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686636/3438343364369609.zip)

adamwilsonCPI, 2020-05-27:
> Hi Adam,
>
> here is the zip files for these device ID, please try it, thanks!!
>
> [343834335E369109.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686629/343834335E369109.zip) [...]

Hi again,

Thanks for these :-) I'll let you know when these have been used, I'm hoping to get them done by the end of the week.

Many thanks,

Adam

lugiaking, 2020-05-27:
Hi Adam,

Thank you for the kindly update!  :)

adamwilsonCPI, 2020-05-28:
> Hi Adam,
>
> here is the zip files for these device ID, please try it, thanks!!
>
> [343834335E369109.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4686629/343834335E369109.zip) [...]

> Hi Adam,
> 
> Thank you for the kindly update! :)

Hi again,

No problem :-) I have a colleague booked in to perform the updates next week.

I have a couple of questions regarding user programming with firmware v4.0.2, if that's okay. 
Every now and again I don't get a "+RCV=..." response after using "AT+SEND", I'm assuming that this operation is timing out. What are the possible responses other than "+RCV=..." that I could get, so I can check for them?
The same for "AT+JOIN=1", what are the possible responses other than "+JoinAccepted" that I could get, so I can check for them?

Many thanks,

Adam

lugiaking, 2020-06-02:
Hi Adam,

the +RCV event only occurs if there is some data on the gateway that needs to be returned to the device after using AT+SEND, if you need to know if data has been sent (onto the gateway) successfully, you can try AT+SEND with the parameter <event_enabled> at the end of command like the example below. and similarly, you also can use the parameter <event_enabled> at the end of command 'AT+JOIN'. this feature only available after firmware version 4.0.2,  you can refer to the page 34 and page 35 from the atcmd manual below, thanks!

https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/WM-SG-SM-42%20AT%20Command%20Reference%20Manual%20rev.2.6_20200225.pdf

for example#1  send out an ACK packet with hex data '0102030405' on the port 2, and report the state for this sending 
# AT+SEND=2,0102030405,1,1

for example#2  join to the gateway with OTAA activation, and report the state for this join 
# AT+JOIN=1,1

adamwilsonCPI, 2020-06-04:
> Hi Adam,
>
> the +RCV event only occurs if there is some data on the gateway that needs to be returned to the device after using AT+SEND, if you need to know if data has been sent (onto the gateway) [...]

Hi again,

Thank you for the extra information :-)

We've managed to get all the other LoRa modems up and running with firmware v4.0.2, which is good news. I've been testing my own hardware and firmware with a v4.0.2 modem, and it has been running reliably for > 48 hrs now, so I believe all my reliability issues are now sorted - thank you!

I'll leave the GitHub issues thread open for the time being, as it seems there are others in need of similar assistance.

Many thanks,

Adam

Tyluuus, 2020-07-15:
Hi,
When I try to connect with I-Nucleo-LRWAN1 shield in ST-Link Utility I've got info: "Can not read memory! Disable Read Out Protection and retry". 

When I try to update shield I've got info: 
[image]

I'd be grateful if you @lugiaking can generate backup files for me, for two devices:

- device id = 3036343259376D02    (Firmware version 2.9)

- device id = 343834336D369508    (Firmware version 2.6)

btw. I've got another problem maybe you will know answer. I implemented simple code based on PingPong for this shield but I can only receive first 4 characters of transmitted message. Could this issue connected with firmware?

Thanks for all answers! 

T.

lugiaking, 2020-07-16:
Hello Tylus,

here are the backup files for the two devices, and regarding to the question about the PingPong demo code, do you have re-start the RX mode from the RX callback? if not so, you just can receive the message once. can you check it again?  thank you!!  

[343834336D369508.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4932356/343834336D369508.zip)
[3036343259376D02.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4932357/3036343259376D02.zip)

Tyluuus, 2020-07-16:
Hi,

I've tried to update board but have this message: 
[image]

I'm using on board ST-Link so i'm disconectting nucleo board, reconnect it and hit Enter but result is shown on screen. You know where can be problem @lugiaking ? 

T.

lugiaking, 2020-07-22:
Hi Tylus,

could you please tell me what firmware you are using for the RF tests? or what test you want to run? I have tried send tx packets by follow the command example in section 4.6 of the document below, and I can see a lot of wave forms appears on the spectrum analyzer.  thanks!!

https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/WM-SG-SM-42%20AT%20Command%20Reference%20Manual%20rev.2.6_20200225.pdf

Tyluuus, 2020-07-22:
Hi @lugiaking ,

I'm using firmware 4.0.2. I'm trying to build project which use trasmitting packets between two boards without LoRaWAN so firstly i'm trying to run program which can send message. I wrote two simple codes: 
[Tx-Rx.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4961633/Tx-Rx.zip)
based on PingPong example and libraries from there: https://github.com/stm32duino/I-NUCLEO-LRWAN1 
[lora.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/4961642/lora.zip) 
Packets are sent very irregular and if only receiving shield can get some data, it only receive first 4 bits. Additionaly idk why this cannot work in loop as written, packets are sent sometimes only if I press reset button on board. 
I also attach result of test: 
[image]

If you have any clue, I'd appreciate.

Kind regards,
Piotr

Tyluuus, 2020-07-22:
Can you also point me best way to print received data by AT commands? I created simple code to send data via AT+TXT command and what's more important it works, but also have problem with receiving. 

Best,
T.

1hwg, 2020-09-26:
Backup.bat will not be performed correctly (sm42 interface seems abnormal ...) if ECHO is enabled in the WM-SG-SM-42 module.
You must prefix backup.bat with the following sequence of AT commands from the terminal.

ATE = 0 / * disable the local echo * /
AT + WDCT / * save changes to eeprom * /

then the module power cycle.

then backup.bat

jgt922, 2020-10-07:
Please Help me. My board's not firmware. **device id = 3036343264376502**
my email is  jgt922@alumnos.unican.es

PS C:\TFG_JOSE\LoraWANServer\I-nucleo-LRWAN1\USI_I-NUCLEO-LRWAN1-master\preloaded_firmware\sm42_fw_update_tool_v1.1-1\bin> .\sm42_update.bat 8 sm42_monolithix_v4.0.2.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v16 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.256353
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 3036343264376502
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!

jgt922, 2020-10-07:
I have a similar problem. Is it possible to get the bak file for the following ID?
**device id = 3036343264376502**
my email is jgt922@alumnos.unican.es
Thanks

bryanstein, 2020-10-28:
I need a new 343834335C369308.bak file. Can you please upload this for me?

I have a similar issue. I backed up my device before upgrading but openocd stackdumps after flashing but never flashes the nvm:

** Programming Finished **
** Resetting Target **
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
shutdown command invoked
      1 [main] openocd-dev 22440 cygwin_exception::open_stackdumpfile: Dumping stack trace to openocd-dev.exe.stackdump

> Hello, I uploaded two 7z files for id '3036343256377403' and '303634327C376602', please try it again, thank you!

lugiaking, 2020-10-29:
Hi jgt922,

here is the recovery file for the device id 3036343264376502, thanks!

[3036343264376502.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5457945/3036343264376502.zip)

lugiaking, 2020-10-29:
Hi bryanstein,

here is the recovery files for all your devices, thanks!!

[303634327C376602.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5457974/303634327C376602.zip)
[343834335C369308.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5457975/343834335C369308.zip)
[3036343256377403.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5457976/3036343256377403.zip)

lugiaking, 2020-10-29:
Hi 1hwg,

thank you for share your experience to everybody, many thanks!!

lugiaking, 2020-10-29:
Hi Tyluuus,

sorry for the late, are you still working on the transmission and reception via AT+TXT command?

Tyluuus, 2020-10-31:
Hi,

I've finished that project successfully. Thanks for asking.

Kind regards

ANauzet, 2020-11-17:
Hi @lugiaking 
I'm trying to backup the firmware in two USI_I-NUCLEO-LRWAN1 module with no succes. Could you please provide me the 2 backup fille for the following module:

DevEUI: 0xe24f43fffe44ccdf
DevEUI: 0xe24f43fffe44d1ce

My email: antoine.nauzet@gmail.com

When trying to backup, I have the error ""sm42 interface seems abnormal..."
I have seens that it might be because of the echo, but I already turn it of. Do you have any idea what cuases this error?

If I try to update to another firmware version, will it work? Or at least not remove the actual firmware if it fail?

Thank you

ugrakm, 2020-11-19:
Please Help me. My board's not firmware. device id = 3438343388367E08
my email is ugura@novalit.com.tr

ugrakm, 2020-11-19:
I need a new 3438343388367E08.bak file. Can you please upload this for me?
my mail is ugrakm@gmail.com
[image]

lugiaking, 2021-02-15:
Hi ANauzet,

sorry for late, here is bak files for you, the UART connection between the host might be abnormal, it will cause the failure to restore the nvm data into the module after done the programming by SWD,  to avoid the problem, I suggest to make sure the AT CMD can be sent and received normally before run the backup or update, many  thanks!

[E24F43FFFE44CCDF.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5979434/E24F43FFFE44CCDF.zip)
[E24F43FFFE44D1CE.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5979435/E24F43FFFE44D1CE.zip)
[E24F43FFFE39CC4B.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5979436/E24F43FFFE39CC4B.zip)
[E24F43FFFE39CB50.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5979437/E24F43FFFE39CB50.zip)
[E24F43FFFE44C45B.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5979438/E24F43FFFE44C45B.zip)

lugiaking, 2021-02-15:
Hi ugrakm,

here is the bak file for you, thanks!!

[3438343388367E08.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5979471/3438343388367E08.zip)

mori-db, 2021-05-12:
Hi @lugiaking

I also need to upgrade the firmware of my end device and I need the bak file.
Info : found unique device id = 3438343378367108

Many thanks,

MBuratto, 2021-09-22:
Hi,
I have the same problem.
Info : found unique device id = 343834335E367308
Thanks

[13 more replies]