USIWP1Module/USI_I-NUCLEO-LRWAN1 #32: Join request DR

aliengravy, 2020-04-18:
Hi all, 

I am having trouble with the join request on EU and the things network.

The device always sends a join request with     
"spreading_factor": 12,
    "bandwidth": 125,

If this fails then there is an incredibly long wait time on the duty cycle.

Is it possible to change the initial join settings?

I have tried just about everything and can't seem to do it. Even disabling channels doesn't seem to work.

aliengravy, 2020-04-27:
I would like to change the SF on a join to a much lower setting to look for very local base stations before sending on SF12 to minimise the duty cycle requirements but cannot find a way.

lugiaking, 2020-05-20:
Hi Aliengravy,

can you tell us what activation mode you are using for the join? ABP or OTAA.
you can try the command below to change the DR before join.

AT+DR
AT+RX2DR

cstratton, 2020-11-12:
> can you tell us what activation mode you are using for the join? ABP or OTAA.

Join requests are only applicable to OTAA

> you can try the command below to change the DR before join.
> 
> AT+DR
> AT+RX2DR

The latter is dubious, Join RX2 data rate must is always the default setting for the applicable LoRaWAN regional parameters, irrespective of the join request data rate, and regardless if a network (such as TTN in Europe) configures a different data rate for non-join RX2.