USIWP1Module/USI_I-NUCLEO-LRWAN1 #34: sm42_fw_update_tool_v1.1-1.zip appear to be corrupted

KiwiBryn, 2020-07-21:
Hi

When I download the sm42_fw_update_tool_v1.1-1.zip file I can't unzip it.

The file is 101K or 106K WInzip (said file was corrupted) and Windows "Extract Compressed (zipped) Folders" said the file was empty.

There appeared to be a couple of links to zip files of update tools of varying sizes.

I would like to upgrade the device so I can use AS923 as I'm based in New Zealand.

When I run AT+VER\r\n on my device it returns 

1.0.1,2.9
OK

# 

Which doesn't match the documentation, is the firmware on my device just really old?

<lrwan_ver>,<fw_ver>,<hw>,<bootloader_ver>

Thanks

Bryn

lugiaking, 2020-07-22:
Hello Bryn,

'1.0.1,2.9' means the firmware version is 2.9 with LoraWAN v1.0.1, and this version did not support AS923 yet, so please upgrade the firmware version to the latest firmware version 4.0.2 by follow the manual below.
I have tried to download the file 'sm42_fw_update_tool_v1.1-1.zip' from the the download link of the update tool below, and then I can unzip the downloaded file and update the fw using this tool, I guess your download may be uncompleted,  can you please try it again? the file size of the zip file is 2,894,481 bytes, I recommend that you can check the file size to see if the download is completed before unzip it.  please let me know if any problems, thanks!!

# download link of update tool:
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/sm42_fw_update_tool_v1.1-1.zip

# download link of update manual:
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20Update%20Preloaded%20AT%20Command%20FW%20Application%20Note%20rev.%201.4.pdf

# the latest firmware using update tool:
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/v4.x/wm-sg-sm-42%20firmware%20v4.0.2/image%20for%20sm42_fw_update_tool/sm42_monolithix_v4.0.2.hex

KiwiBryn, 2020-07-22:
Sorry to have troubled you, I changed browser and can now unzip the download.

Will flash devices tomorrow and write some libraries for your module.

Thanks again

closed 2020-07-22 by KiwiBryn