USIWP1Module/USI_I-NUCLEO-LRWAN1 #37: Problem with sensors HTS221

Tyluuus, 2020-08-16:
Hi, 

I'm using Nucleo L073RZ with I-Nucleo-LRWAN1 module. I'm trying to get temperature and humidity value but it sometimes work and other times no. 

I'm using "HTS221Sensor" and have quite simillar code like this from Issue #5 (https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/issues/5) (based on example from internet). I can't find a pattern, it can read value but often they are very strange, like humidity 6553,5 rH%, or temperature -0,1 C. It can read correct value but for example after restart from button it cannot work as it should.

Help will be very appreciated.
Thanks in advance!
T.

Tyluuus, 2020-08-16:
After it wait for some time it can read correct value once and get crazy again.

Tyluuus, 2020-08-17:
I also checked result of HTS221_IsMeasurementCompleted function and if after starting a program it keep failinig untill restart.

AdamMax92, 2022-01-17:
Hello Tyluuus

I am trying to read Data from the HTS221 via I2C. I am struggling a little due to the lack of documentation.
Could you maybe post your I2C functions to check where my errors are?