USIWP1Module/USI_I-NUCLEO-LRWAN1 #39: Cannot update firmware

jgt922, 2020-10-08:
Hello,

I am trying to update firmware (following the document https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20Update%20Preloaded%20AT%20Command%20FW%20Application%20Note%20rev.%201.4.pdf) , but whatever I try, I get this

My problem equals # 9. You can send me the .bak file for this device id: **3036343264376502**
My email jgt922@alumnos.unican.es
Thanks greetings.

PS C:\TFG_JOSE\LoraWANServer\I-nucleo-LRWAN1\USI_I-NUCLEO-LRWAN1-master\preloaded_firmware\sm42_fw_update_tool_v1.1-1\bin> .\sm42_update.bat 8 sm42_monolithix_v4.0.2.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 2
going to run using 'stlink_v2-1'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v16 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.256353
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 3036343264376502
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!

lugiaking, 2020-10-29:
Hi jgt922,

here you are, thank you!

[3036343264376502.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5457882/3036343264376502.zip)

lugiaking, 2020-10-29:
Hi rony,

here is the recovery file for device id 303634326B377403, thanks!

[303634326B377403.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5457886/303634326B377403.zip)

ANauzet, 2020-11-26:
Hello,
I'm asking for help here:
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/issues/9
and here
https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/issues/30

9 days ago, and still no responses from USI support.
I need bak file to unblock my module

My adresse: antoine.nauzet AT gmail.com

DevEUI: 0xe24f43fffe44ccdf
DevEUI: 0xe24f43fffe44d1ce

Thank you

Jezek-David, 2021-02-07:
Hello i have a same problem:

C:\Users\David\Desktop\sm42_fw_update_tool_v1.1-1\sm42_fw_update_tool_v1.1-1\bin>sm42_update.bat 6 sm42_monolithix_v4.0.2.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v37 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.147668
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = 303634326F379D03
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!

UweBonnes, 2022-09-17:
One error here i.m.h.o. is that it must be comX versus the pure X argument!