USIWP1Module/USI_I-NUCLEO-LRWAN1 #40: USI_I-NUCLEO-LRWAN1 with STM32NUCLEO

Jezek-David, 2021-02-01:
Hello, my project is focused on sending data using lora. I have a Nucleo-STM32F767ZI at home. I've initialized UART6 and I'm trying to send commands to I-NUCLEO-LRWAN1. But I don't get any answers from the board. The question is, is it possible to put this into operation. Only using AT commands?

stomascik, 2021-02-01:
Sure. https://twitter.com/SlavoTomascik/status/1349508003054186497

Jezek-David, 2021-02-01:
Nice, do you have any example in what format you send commands? Or some specific example of setting.

stomascik, 2021-02-01:
I use commands like in the WM-SG-SM-42 AT Command Reference Manual rev.2.6_20200225.pdf chapter 4.11.1. Almost exactly. Commands send by terminal by br@y. But any other terminal should be ok. Sometimes i use "serial USB Terminal" on android tablet.

stomascik, 2021-02-01:
So, i recommend to start with ABP. I am stuck for many hours, because OTA do not work for me. Probably because of low signal levels. Device many times retry for success connection.

closed 2021-02-07 by Jezek-David