USIWP1Module/USI_I-NUCLEO-LRWAN1 #41: I-NUCLEO_LRWAN1 update frimware

Jezek-David, 2021-02-07:
Hello, i have a problem again. I can't update the firmware this time. I have: AT + VER1.0.1,2.9. I have ST-Link and UART connected to my computer. In ST-Link Utility, I connect to the device and change the settings in Option bytes according to: https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20recover%20guide.pdf
But when I try to backup via cmd I only get here:
C:\Users\David>C:\Users\David\Desktop\LORA\USI_I-NUCLEO-LRWAN1-master\USI_I-NUCLEO-LRWAN1-master\preloaded_firmware\sm42_fw_update_tool_v1.1-1\sm42_fw_update_tool_v1.1-1\bin\sm42_backup.bat 6
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
'openocd-dev' is not recognized as an internal or external command,
operable program or batch file.
I will be very happy for the advice.

stomascik, 2021-02-07:
Why do you try to recover firmware? Did you test firmware update from https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/WM-SG-SM-42%20Update%20Preloaded%20AT%20Command%20FW%20Application%20Note%20rev.%201.4.pdf
Binary is https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/blob/master/preloaded_firmware/sm42_fw_update_tool_v1.1-1.zip

Jezek-David, 2021-02-07:
I tried both. The module responds to AT commands as it should. But when I mow the down payment, I always get a message: C:\Users\David>C:\Users\David\Desktop\sm42_fw_update_tool_v1.1-1\sm42_fw_update_tool_v1.1-1\bin\sm42_backup.bat 6
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
'openocd-dev' is not recognized as an internal or external command,
operable program or batch file.

stomascik, 2021-02-07:
Seems, you run sm42_backup.bat from C:\Users\David directory. Open command line in the C:\Users\David\Desktop\sm42_fw_update_tool_v1.1-1\sm42_fw_update_tool_v1.1-1\bin\ and run from this place

Jezek-David, 2021-02-07:
Thanks, that helped. My fault. Now I have this error:
C:\Users\David\Desktop\sm42_fw_update_tool_v1.1-1\sm42_fw_update_tool_v1.1-1\bin>sm42_backup.bat 6
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...

stomascik, 2021-02-07:
Sorry, no idea why this error. Do you really have a stlink_v2?

Jezek-David, 2021-02-07:
Yes, with FW: V237S7

stomascik, 2021-02-07:
It's weird. I have saved my update process (not a backup) and there is same error, but update continue successful.
Try to check connection UART and SWD.

Part of my log:
d:\Documents\Datasheets\USI\sm42_fw_update_tool_v1.1-1\bin>sm42_update.bat 4 sm42_monolithix_v4.0.2.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v37 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.326453
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
...

Jezek-David, 2021-02-07:
Now I get this:

C:\Users\David\Desktop\sm42_fw_update_tool_v1.1-1\sm42_fw_update_tool_v1.1-1\bin>sm42_update.bat 6 sm42_monolithix_v4.0.2.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
/dev/ttyS5: Permission denied

stomascik, 2021-02-07:
Again, no idea. Try to you run command line as administrator?

Jezek-David, 2021-02-07:
Now I get this: 

C:\Users\David\Desktop\sm42_fw_update_tool_v1.1-1\sm42_fw_update_tool_v1.1-1\bin>sm42_update.bat 6 sm42_monolithix_v4.0.2.hex
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...
Info : target responses abnormal ...
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v37 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.147185
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : STM32L flash size is 64kb, base address is 0x8000000
Info : found unique device id = ......
Info : atcmd responses abnormal, will use previous nvm backup
Error: Ohh, cannot find previous nvm backup, this unit cannot be UPDATE and RECOVERY by this tool !!!

stomascik, 2021-02-07:
It's promising. Run sm42_backup.bat first for create backup file.

Jezek-David, 2021-02-07:
I have two modules available, one managed to update the other, I can't even make a backup: he still writes: 
C:\Users\David\Desktop\sm42_fw_update_tool_v1.1-1\sm42_fw_update_tool_v1.1-1\bin>sm42_backup.bat 6
"0) stlink-v1"
"1) stlink_v2"
"2) stlink_v2-1"
select a profile no. from above? 1
going to run using 'stlink_v2'
Open On-Chip Debugger 0.9.0-dev-v1.2 (2017-02-27-12:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: sm42 interface seems abnormal ...

Jezek-David, 2021-02-08:
I have another problem. Connecting to the gateway. The JOIN request arrives and the gateway sends the receipt, but nothing is displayed on the module communication and the module sends the request again.

[image]

ERROR_WAN_NON_JOINED

lugiaking, 2021-02-09:
sorry for my late, I made some backup as below for the recovery, 

[E24F43FFFE44CCDF.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5952569/E24F43FFFE44CCDF.zip)
[E24F43FFFE39CC4B.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5952571/E24F43FFFE39CC4B.zip)
[E24F43FFFE39CB50.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5952573/E24F43FFFE39CB50.zip)
[E24F43FFFE44C45B.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5952574/E24F43FFFE44C45B.zip)
[E24F43FFFE44D1CE.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5952575/E24F43FFFE44D1CE.zip)

please put them in the path 'sm42_fw_update_tool_v1.1\bin' and please help to have a try with command below, thanks!
:/> sm42_update.bat <com_port_number>  <hex_file>

lugiaking, 2021-02-09:
Hi David,

regarding to another problem about the ERROR_WAN_NON_JOINED,  since SF7 has the shortest communication distance, could you please shorten the distance and try it again ... I am not sure how far is the best distance for SF7,  can you please help to try APB with another data rate?  many thanks!

Jezek-David, 2021-02-10:
Thanks, but it doesn't work:
Error: invalid backup data (1) - for all .bak file
My unique device id = 303634326F379D03
Thanks.

lugiaking, 2021-02-15:
Hi David,

sorry, here is the bak file for unique id '303634326F379D03'  thanks!

[303634326F379D03.zip](https://github.com/USIWP1Module/USI_I-NUCLEO-LRWAN1/files/5979326/303634326F379D03.zip)

Jezek-David, 2021-02-17:
So I updated the module. But I still have a problem with connecting to the gateway. OTAA does not work which after entering AT + JOIN = 1 I will not receive an acknowledgment (+ JoinAccepted). APB also does not work after setting everything the gateway will not accept anything at all. Distance is not a problem I tried it at a distance of 4m 10m and 20m still the same results.

lugiaking, 2021-02-18:
Hi David,

sorry, so is the traffic log on the gateway the same as you posted before?  the join-accpet has been sent from GW and module always miss?  is the region setting between module and GW the same?

Jezek-David, 2021-02-19:
Hello, here is my gateway settings.
[image]

[image]

I'm still having trouble getting the USI module up and running anyway. I reset everything to SF9. But I still don't get a hint that the gateway would accept anything.

[image]

Jezek-David, 2021-02-23:
Also when I run the test mode on the module. (DEFMODE = 2). The gateway shows this:

(RXPKT): [up] {"rxpk":[{"time":"2021-02-23T16:29:08.355886Z","tmst":1793878892,"chan":0,"rfch":1,"freq":868.000000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.0,"rssi":-68,"size":1,"data":"OA=="}]}
 (RXPKT): [up] {"rxpk":[{"time":"2021-02-23T16:29:08.366454Z","tmst":1793889460,"chan":0,"rfch":1,"freq":868.000000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":8.0,"rssi":-68,"size":1,"data":"pw=="}]}
 Receive(HEX):58
 (RXPKT): [up] {"rxpk":[{"time":"2021-02-23T16:29:08.541746Z","tmst":1794064755,"chan":0,"rfch":1,"freq":868.000000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":8.0,"rssi":-68,"size":1,"data":"WA=="}]}
 Receive(HEX):7a
 Receive(HEX):bb
 (RXPKT): [up] {"rxpk":[{"time":"2021-02-23T16:29:08.634859Z","tmst":1794157852,"chan":0,"rfch":1,"freq":868.000000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":8.0,"rssi":-68,"size":1,"data":"eg=="}]}
 (RXPKT): [up] {"rxpk":[{"time":"2021-02-23T16:29:08.645465Z","tmst":1794168460,"chan":0,"rfch":1,"freq":868.000000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.0,"rssi":-68,"size":1,"data":"uw=="}]}
..........

closed 2021-03-02 by Jezek-David